letscontrolit/ESPEasy #4603: Can't find the normal 1M version to flash my esp8266-01s

komega13, 2023-04-12:
Hi,
I follow tuto as I'm newbie to flash my esp8266 with ESPeasy, every tuto says that I need to use the "ESP_Easy_mega_20230409_normal_ESP8266_1M.bin"
But I can't find it, even the online version of the flasher that propose to me this bin is unable to find and download it.
Wich .bin should I use instead please ?

TD-er, 2023-04-12:
The build script does filter out build files which end up too large.
If you flash those, the file system area will get partly overwritten.

I just tried building it on my PC and it is indeed about 700 bytes too large.

I will have a look at this to see what can be left out.

Which plugins and controllers do you need?

komega13, 2023-04-12:
I'm not really sure what you mean by plugins and controller but I want to use my ESP8266-01S with a l298n to control a linear actuator via Jeedom (I have dowload ESPeasy plugin on Jeedom market)

TD-er, 2023-04-12:
OK, so you will compile your own ESPEasy build?

A plugin is a piece of code for ESPEasy to interact with some hardware like a sensor, display, etc.
A controller is a piece of code for ESPEasy to send data (e.g. sensor samples) to some other host (e.g. MQTT broker)

ESPEasy builds differ mainly in the included plugins and controllers.
On top of that, there are some differences like intended CPU (ESP32/ESP8266) and flash size.
These differences in plugin combinations is needed, since these plugins all add up to the build size and thus a build with all plugins will not fit on smaller flash sizes.

Since there currently isn't an 1M build in the latest release, I can build one for you, but then I need to know which plugins and or controllers you need.

komega13, 2023-04-12:
I guess a normal 1M build is what I need, the code I will put via ESPeasy is just a ON OFF and REVERSE for the linear actuator. The rest of the interactions (sensor etc...) will be done via Jeedom plugin

TD-er, 2023-04-12:
Do you have a link to this "Jeedom plugin" ?
The word "plugin" is a bit unclear to me here, as we use this word in ESPEasy for a piece of code which needs to be compiled into ESPEasy.

komega13, 2023-04-12:
Jeedom plugins are avaible on the Jeedom platform market, it's difficult to send you a link of this without having Jeedom . Plugins are used to make a bridge between different protocol and the web interface, for exemple I used a Z-wave plugin to make Jeedom communicate with all my Z-wave protocol based in my house. 

The plugin is develloped by lunarok, here the page of the documentation (in French) https://lunarok.github.io/jeedom_docs/plugins/espeasy/

But I'm pretty sure you can ignore what the plugin do because all people who used it doesn't have a specific firmware, just normal firmware

TD-er, 2023-04-12:
Hmm they do refer to the really outdated documentation on the ESPEasy wiki.
The latest documentation for ESPEasy is here: https://espeasy.readthedocs.io/en/latest/

Anyway, there is literally nothing stated on that page what is needed on the ESPEasy side.
I can't see which controller should be included in ESPEasy, nor what plugins are needed.

So this makes it really hard for me to make a build that will fit on the really small 1M flash.

Maybe if you have some ESP8266 board with 4M, you can at least test a bit to see what Jeedom does do with an ESP board running ESPEasy. Then we can tomorrow see what plugins and controllers you need so I can make a build for you that will fit on an 1M flash unit.

The "minimal" 1M builds should also work just fine, but I am not sure those will have what's needed by Jeedom.
So if you have some 4M ESP board laying around, you can at least test what Jeedom needs, even though that board doesn't have the hardware connected to control/read your linear actuator.

komega13, 2023-04-12:
Ok thanks for your support you are very nice.
I'm amazed concerning the 1M firmware because the ESP8266 with 1M memory is one of the most famous chip on Youtube DIY channel were I got inspired for many project.
The ESP_Easy_mega_20230409_normal_ESP8266_1M.bin existed and I don't understand why he was removed

Here the link of the video (in french again) were the guys use an ESP8266 with ESPeasy to command a piston maybe it can help you a bit : https://www.youtube.com/watch?v=GTrCb264280&t=457s

TD-er, 2023-04-13:
> The ESP_Easy_mega_20230409_normal_ESP8266_1M.bin existed and I don't understand why he was removed

The flash is partitioned.
- bootloader
- sketch
- room for OTA update
- file system

The "sketch" and "OTA update" area are shared on ESP8266.
Meaning that you can have such a large sketch that there is no room for OTA update anymore and you only can update via serial.
However the sketch should never be so big that it will overwrite the file system as this will corrupt settings and the other way around if the file system is created it will overwrite part of the sketch binary.

Now the "build problem"...
When building the image, the build tools only check for the size of the "sketch" and not the final bin size which includes the bootloader stuff (and some other stuff, but just for simplicity let's keep with the "bootloader" name)
So I added an extra check in the build process that does check the final bin size and if it won't fit, the file will be copied to the "rejected" folder instead of the "build_output/bin" folder.

So there never was a release of the `ESP_Easy_mega_20230409_normal_ESP8266_1M.bin` file as it was deemed too large after building it.

Thus the only thing I can do is leave out some plugins/controllers to help make it small enough for you to flash.

ThibaultLef, 2023-05-05:
@TD-er indeed since the mega-20230409 release, binaries have disappeared. 
By trying the binaries `ESP_Easy_mega_20230409_custom_ESP8266_1M.bin` the generic MQTT import function seems to be missing. 
Is there anything you can do solve it? Thank you

tonhuisman, 2023-05-06:
@ThibaultLef I have created a PR (link above this message) to disable a few plugins (NeoPixel related) in the ESP8266 1M builds. You can download from [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4900487065) for testing.

PR #4648 (2023-05-07): [Build] Make 1M builds fit by excluding some plugins
Resolves #4603 

Feature:
- Exclude NeoPixel Word Clock, Candle and Clock from ESP8266 1M builds (except from the NeoPixel 1M builds)
closed 2023-05-07 by TD-er