letscontrolit/ESPEasy #4604: No countdown when connecting to my SSID

komega13, 2023-04-12:
After flashing the ESP8266-01S it finaly appear on my WIFI network. I connect whith my smartphone and entering the default password "configesp" then I open the 192.168.4.1 web page in my smartphone browser and the ESP easy configuration web page appear.
Then I select my home SSID and type my password. Then nothing. No countdown the page is empty. 

The ESP8266 is not showing up anymore on my WIFI and nothing new is connected to my router. Any idea ?

TD-er, 2023-04-12:
Which build did you flash?
In the other issue, you reported there wasn't a 'normal 1M' build.

TD-er, 2023-04-12:
You can also try via a terminal program (and thus via the serial port) to enter the following commands:

```
wifissid,YourNetworkName
wifikey,YourSecretWiFiPassword
save
reboot
```

This way the ESP does not need to have both the AP mode active and try to connect to WiFi.
With AP mode active, the ESP may need just a bit more power and if the voltage regulator (or the cables) are not up for this, you may end up with all kinds of strange issues like partly saved settings for example.

komega13, 2023-04-12:
I try all the build that have "1M" in their name, the last I use was the domoticz one

komega13, 2023-04-12:
I'm pretty newbie I'm not sure wich terminal to use and how to communicate with the COM serial port, is this cmd.exe in windows 10 that I should use ?

TD-er, 2023-04-12:
When using the web flash tool, you can also open a "console" instead of flashing.
In this console window, you will see all kinds of logs and you can also enter commands, like I wrote in my previous post.
You only need to replace the WiFi SSID and network key with your own.

komega13, 2023-04-12:
I have directly enter the SSIS and the password in the easy flasher (using the domoticz one firmware for testing), when I open serial monitor it seems that the ESP is connected and having it's own IP adresse if i read good. Here a screenshot
[image]

komega13, 2023-04-12:
[image]

TD-er, 2023-04-12:
Ah good to see it is now working.
That's at least one step closer to the end goal :)

fly74, 2023-04-14:
@TD-er What is the command to connect to a hidden Wi-Fi?

 > You can also try via a terminal program (and thus via the serial port) to enter the following commands:
> ```
>
> wifissid,YourNetworkName
>
> wifikey,YourSecretWiFiPassword
>
> save
>
> reboot
>
> ```
>
>
>
> This way the ESP does not need to have both the AP mode active and try to connect to WiFi.
>
> With AP mode active, [...]

@

TD-er, 2023-04-14:
Hmm I see what you mean...
`Include Hidden SSID:` is unchecked by default and there is no command for enabling it.

That's for sure a valid feature request :)

fly74, 2023-04-14:
Can we make it default?

TD-er, 2023-04-14:
> Can we make it default?

Made an issue for it: https://github.com/letscontrolit/ESPEasy/issues/4609

Not sure if that's the best option as it really makes connecting to WiFi taking a lot longer when the configured SSID isn't found immediately and having a hidden SSID isn't that common.