letscontrolit/ESPEasy #4605: [Feature] Add PutToHTTP command for HTTP PUT

BasvanH, 2023-04-13:
Following up on #4446 I would like to request to add HTTP PUT method to be used in rules. This again to enhance support for [OpenHAB](https://www.openhab.org/docs/configuration/restdocs.html) API which requires the PUT method to update an item. Other API integrations would also benefit from adding PUT to the rules engine.

A lot of the logic/code of POST can be reused to support PUT so I think and hope this will be a relative easy feature to add.

tonhuisman, 2023-04-17:
@BasvanH I've opened a PR that adds the `PutToHTTP` command, you can test by downloading a build from [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4724377190), once that's finished building.
Syntax is _very_ similar to the `PostToHTTP` command 😉

BasvanH, 2023-04-20:
Tested, all good. Much appreciated!

PR #4619 (2023-04-21): [Commands] Add PutToHTTP command
Resolves #4605 

Features:
- Add command `PutToHTTP`, very similar to `PostToHTTP`, but using the `PUT` http verb.
- Add documentation for `PutToHTTP`.

TODO:
- [x] Testing (by requester) ([confirmed](https://github.com/letscontrolit/ESPEasy/pull/4619#issuecomment-1515772736))
closed 2023-04-21 by TD-er