letscontrolit/ESPEasy #4608: TaskValueSet command and number over 8 characters

SirWant, 2023-04-13:
TaskValueSet command is not valid with numbers over 8 characters (for example, Timestamp):

```
On test Do
  TaskValueSet,Mute,MuteTime,1234567890
  LogEntry,[mute#mutetime]
  TaskValueSet,Mute,MuteTime,123456789
  LogEntry,[mute#mutetime]
  TaskValueSet,Mute,MuteTime,12345678
  LogEntry,[mute#mutetime]
Endon
``` 

23819863: EVENT: test
23819866: ACT  : TaskValueSet,Mute,MuteTime,1234567890
23819873: ACT  : LogEntry,1234567936
23819874: 1234567936
23819875: ACT  : TaskValueSet,Mute,MuteTime,123456789
23819881: ACT  : LogEntry,123456792
23819882: 123456792
23819884: ACT  : TaskValueSet,Mute,MuteTime,12345678
23819889: ACT  : LogEntry,12345678
23819890: 12345678

Actual Custom Build ESP8266_4M1M

TD-er, 2023-04-13:
That's completely according to how 32 bit float will behave :)
Task values are kept as a `float`.
If you need to store longer integer types, upto 32 bit, you can store it in a dummy task which is set to be of type "long".
When you need to store more bits, you can store it in rules in variables.
Those are of type `double`.

SirWant, 2023-04-14:
> you can store it in a dummy task which is set to be of type "long"

Do I need to convert somehow?

1784243: EVENT: test
1784247: ACT  : TaskValueSet,Mute,MuteTime,1234567890
1784257: ACT  : LogEntry,3936486144
1784258: 3936486144
1784259: ACT  : TaskValueSet,Mute,MuteTime,123456789
1784265: ACT  : LogEntry,2825375000
1784266: 2825375000
1784267: ACT  : TaskValueSet,Mute,MuteTime,12345678
1784272: ACT  : LogEntry,2714263886
1784274: 2714263886

> you can store it in rules in variables

I need to store the time as a setting.

TD-er, 2023-04-14:
Something like this should work:

[image]

SirWant, 2023-04-14:
I did it, the log is above.

TD-er, 2023-04-14:
Just to check it isn't a JavaScript update issue, I also tried this:
```
logentry,"setDummyValue: [dummy#unixtime]"
```
with this log as result:
```
266085648: HTTP: logentry,'setDummyValue: [dummy#unixtime]'
266085656: setDummyValue: 1234567936
```

TD-er, 2023-04-14:
I was already working on adding extra "dummy types" in this PR: https://github.com/letscontrolit/ESPEasy/pull/4533
But that's not yet finished.
I guess this issue is actually a perfect description to why PR is needed :)

So thank you to help find the problem for the solution I was already working on :)

SirWant, 2023-04-14:
Okay, we're waiting, it'll be wonderful :)

TD-er, 2023-04-14:
OK, just tested with that PR and got this:
[image]

So that's already working :)

SirWant, 2023-04-14:
> Long Quad

Yes, I need a lot of them :)

> type `double`?

It is possible, but also not sure that large foat values ​​are important to someone...

TD-er, 2023-04-14:
> It is possible, but also not sure that large foat values ​​are important to someone...

When working with GPS-coordinates, the resolution in float becomes challenging when you're at higher longitude degrees, or closer to the poles.

I'll just add them as the internal rules calculations are already done in double.

SirWant, 2023-04-15:
Are there any other plans for the ability to save values ​​in case of power loss? Or user settings with saving user variables in to user_config.dat?

TD-er, 2023-04-15:
Plans: yes
Planned ETA: Not yet.

PR #4533 (2023-04-16): [Dummy] Add 1...4 ULong as taskvalue output type
This way we can store upto 4  `uint32_t` values in a dummy and hand them over to the Cache Controller.

Fixes: #4608
closed 2023-04-16 by TD-er