letscontrolit/ESPEasy #4610: Last Firmware 2023 with PZEM004 V3  plugin for ESP8266 4M1M

Ricco59163, 2023-04-14:
Hi all,

who can link the last firmware 2023  with PZEM004 V3  plugin for ESP8266 4M1M ?

if I use the Max Version , I must buy new ESP8266 with 16M

Thx for Help

Ricco59163, 2023-04-15:
When Upload ESP_Easy_mega_20230409_energy_ESP8266_4M1M.bin  in devices  "  Plugin  P251 not included in build "

Ricco59163, 2023-04-15:
Solved , plugin in not same place of 2019 version

thx for help

tonhuisman, 2023-04-15:
That P251 originates from the ESPEasyPluginPlayground, that hosts plugins that are unsupported, even though some are working perfectly fine. Some plugins may or may not have been migrated to the official ESPEasy mega branch, but the code isn't removed from the playground, as someone may still be using it.
The PZEM plugin has indeed been migrated some years ago, but did get a different ID, as these IDs have to be unique, and we assing new IDs in sequence. There is no conversion of settings possible, as there are possibly/probably other experimental plugins that use the same ID.

As this issue is resolved you can now close it 👍

closed 2023-04-15 by Ricco59163