letscontrolit/ESPEasy #4611: MQTT - Publish Retain Flag per Device

iz8mbw, 2023-04-15:
Hello.
For what I know now the MQTT Publish Retain Flag is a "global" setting in Controller, so this means that all the Devices will adopt this setting.
It will possible to specify the Publish Retain Flag as a setting per Device?
In this way we can choose which Device should use the Publish Retain Flag when publish MQTT messages.

Thanks.

TD-er, 2023-04-15:
Would it be "good enough" to have the retain flag as an argument (or a separate "publish" command) to be used in the rules?

TD-er, 2023-04-15:
I did have a look at the setting struct.
```c++
boolean TaskDeviceSendData[CONTROLLER_MAX][N_TASKS];
```
This only holds the checkbox info per task to which controller something should be sent.
However due to the way how C++ does organize structs in memory, we can change this into an `uint8_t` and the rest of the struct remains the same (nothing is shifted)
This means we could add extra bits here per task, like for MQTT controllers the "retain" bit.

The reason I'm looking into this now, is that the Domoticz IDX is also a task specific config for Domoticz controllers.
And it makes sense to have task specific things configured on the task page...

Still have to look into the rest of the MQTT code to see what flags can be set per message and which need to be set while making the connection.

iz8mbw, 2023-04-15:
To set/configure the Publish Retain Flag per Device in the Web interface is much better then to "play" with Rules.

TD-er, 2023-04-15:
Well I guess the rules publish command should also have such an option as it makes no sense to allow it per task and not via rules

iz8mbw, 2023-04-23:
Hi.
To have the Retain Flag per Task will be PERFECT.
I hope it can be implemented.
If you need help for testing I'm available.
Many thanks.

tonhuisman, 2025-08-09:
@iz8mbw If you are available, I've implemented this setting in a PR, linked above. Can you please test this? I can provide a build, if needed.

iz8mbw, 2025-08-09:
@tonhuisman Hi Ton.
I'm out for holidays, so I'm sorry but I can test on September.
Thanks

iz8mbw, 2025-08-10:
And yes, one per year is mandatory. :-)

PR #5371 (2025-09-07): [MQTT] Add Retain option per task
Resolves #4611 
Resolves #1718 

Features:
- Add a Retained flag for each MQTT Controller per task

The Retained option sends data with `Retain` enabled for the task. this setting is overridden by the Controller **Publish Retain Flag** setting, then it will send the Retain flag for _each_ task's value sent.

TODO:
- [ ] Testing
- [x] Update documentation
closed 2025-09-07 by TD-er