letscontrolit/ESPEasy #4612: BME680 - Wrong GAS values

iz8mbw, 2023-04-15:
Hello.
I suppose the GAS values of the sensor BME680 returned by ESPEasy are wrong.

This is what I see in ESPEasy:
[image]

This is what I see for the same BME680 sensor connected to a Raspberry Pi using [this library/code](https://github.com/pimoroni/bme680-python)
```
read-all.py - Displays temperature, pressure, humidity, and gas.

Press Ctrl+C to exit!

Calibration data:
par_gh1: -46
par_gh2: -3446
par_gh3: 18
par_h1: 615
par_h2: 1040
par_h3: 0
par_h4: 45
par_h5: 20
par_h6: 120
par_h7: -100
par_p1: 37346
par_p10: 30
par_p2: -10361
par_p3: 88
par_p4: 9459
par_p5: -151
par_p6: 30
par_p7: 52
par_p8: -3936
par_p9: -2068
par_t1: 25723
par_t2: 26433
par_t3: 3
range_sw_err: 2
res_heat_range: 1
res_heat_val: 52
t_fine: 105756

Initial reading:
gas_index: 0
gas_resistance: 5654450.814649006
heat_stable: False
humidity: 60.648
meas_index: 0
pressure: 1003.47
status: 32
temperature: 20.66

Polling:
20.66 C,1003.50 hPa,60.66 %RH
20.69 C,1003.49 hPa,60.64 %RH,8555.959041753495 Ohms
20.75 C,1003.47 hPa,60.56 %RH,10326.980212585338 Ohms
20.82 C,1003.49 hPa,60.44 %RH,11576.19701521771 Ohms
20.89 C,1003.47 hPa,60.30 %RH,12533.922403824885 Ohms
20.95 C,1003.50 hPa,60.13 %RH,13283.717839117897 Ohms
21.01 C,1003.47 hPa,59.95 %RH,13877.967070167282 Ohms
21.06 C,1003.49 hPa,59.80 %RH,14369.539483416804 Ohms
21.11 C,1003.50 hPa,59.65 %RH,14876.205202481015 Ohms
21.16 C,1003.49 hPa,59.50 %RH,15121.467730898134 Ohms
```

It seems ESPEasy is using the `Adafruit_BME680 Library v1.0.5 (https://github.com/adafruit/Adafruit_BME680/tree/1.0.5)` according to what I read here https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P106_BME680.ino (very old library).

Currently the latest Adafruit library for BME680 is version 2.0.2 (https://github.com/adafruit/Adafruit_BME680).
May be it should be updated to obtain correct GAS values?

Thanks.

iz8mbw, 2023-04-15:
@tonhuisman can you check when you can?
Thanks.

TD-er, 2023-04-15:
Also keep in mind when comparing newer/updated libraries that we cannot use precompiled binary blobs from Bosch.
When this sensor was released, Bosch had a precompiled library made available for various platforms and SDK versions with their proprietary air-quality-index code.
Due to the license of that additional code, we cannot use that part.

I have no idea whether these restrictions still apply, but make sure to check this when evaluating any updated code for this.

iz8mbw, 2023-04-15:
but for example Adafruit and Pimoroni are using the latest Library and I suppose are free to use.

tonhuisman, 2023-04-15:
In the PR I just created I've updated to the latest available Adafruit BME680 library, but the only calculation that's done on the Gas value is dividing it by 1000 to get a float value, no special processing by using the Bosch code, as that's not included (as stated before). When providing enough decimals (at least 3) you can recreate the raw value, and do any desired processing external from ESPEasy.

iz8mbw, 2023-04-15:
Ok, so if I understood correctly the GAS values returned by ESPEasy are divided by 1000 so to have the "correct" values is enough to multiply per 1000?

tonhuisman, 2023-04-16:
Yes, the value returned from the Adafruit library is an uint32_t holding the MOX resistance in Ohm, and ESPEasy divides that by 1000, as shown in the examples that are included with that library. That method is most likely used to avoid overflow of the float value for (very) high resistance results.

iz8mbw, 2023-04-16:
OK, thanks.
Going to test the [PR you created](https://github.com/letscontrolit/ESPEasy/pull/4615).

iz8mbw, 2023-04-16:
OK, just tested [this Action run](https://github.com/letscontrolit/ESPEasy/actions/runs/4709589034/) about your PR and all works well, here the results with the Formula "%value%*1000".

[image]

Thanks, for me you can merge the PR into Master.

More, since BME680 is software compatible with the latest chip BME688, I suggest to rename in ESPEasy "BME680" to "BME68x".

Thanks!

TD-er, 2023-04-16:
> Yes, the value returned from the Adafruit library is an uint32_t holding the MOX resistance in Ohm, and ESPEasy divides that by 1000, as shown in the examples that are included with that library. [...]

I have no idea why the value is divided by 1000.
But for sure it does not make sense to do it to prevent overflows.
A float can hold way larger values compared to  any type of int.

I think if this returned value actually represents a resistance value, then it for sure needs further processing.
But representing it as kOhm does make sense, if it is actually referring to some resistance.

iz8mbw, 2023-04-16:
For how the code is now the returned GAS values is in kOhm

tonhuisman, 2023-04-16:
I'm adding a setting to choose between Ohm and kOhm output for `Gas`, as well as renaming to BME68x and adding documentation.

tonhuisman, 2023-04-16:
New build will be available from [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4713297238).

iz8mbw, 2023-04-16:
Failed to build or should still wait,?

TD-er, 2023-04-16:
Wait for this one (almost done): https://github.com/letscontrolit/ESPEasy/actions/runs/4713433102

iz8mbw, 2023-04-16:
OK, tested [this](https://github.com/letscontrolit/ESPEasy/actions/runs/4713433102) and with
"Present `Gas` in Ohm (not kOhm) enabled":
[image]

"Present `Gas` in Ohm (not kOhm) dosabled":
[image]

Renamed OK:
[image]

Just a question, with Raspberry and Pimoroni library I was able to get values in Ohm with decimals (thousands with decimals), here the decimals are lost (are always 00) if I set Ohm.
No chance to have decimals if value is set to Ohm?

TD-er, 2023-04-16:
You can't set the decimals in the task config?

iz8mbw, 2023-04-16:
I try to explain myself better.
In the Device configuration for the BME680 I have (by default) two decimals:
[image]

but the decimals are always "00" if I set to Present GAS in Ohm, it looks like the variable inside ESPEasy don't have enough characters (numbers) to contain also decimals.
In fact all the sensor reading never have decimals:

[image]
[image]
[image]
[image]

as you can see, decimals are always "00".

TD-er, 2023-04-16:
As Ton mentioned, it is initially an integer value when read from the library.

iz8mbw, 2023-04-16:
OK, thanks.

tonhuisman, 2023-04-16:
Thanks for the feedback 👍 

Decimals for Gas can indeed be set to 0 if you have Ohms selected. Can't do that automatically, as with that setting unchecked, these decimals _are_ of value, and I don't want to mess up what the user may have configured. Some ppl like to have the numbers nicely aligned (all having 2 decimals).

iz8mbw, 2023-04-16:
Thanks to you for your work!
Two decimals by default in ESPEasy are perfect!

PR #4615 (2023-04-21): [P106] Update BME680 library to latest, rename to BME68x
Resolves #4612 

Features:
- Update Adafruit BME680 library to latest available
- Fix typo in #ifdef
- Rename to `Environment - BME68x` as the BME688 sensor is also supported (backward compatible with BME680), though we do not support the additional AI-features.
- Add setting to present `Gas` in Ohm instead of kOhm.
- Add documentation.

TODO:
- [x] Testing ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/4612#issuecomment-1510385177))
closed 2023-04-21 by TD-er