letscontrolit/ESPEasy #4614: Support for ESP32-S3

tomatos-st, 2023-04-15:
Are you planning to add support for ESP32-S3?
I bought ESP32-S3-DevKit C N16R8 [https://www.aliexpress.com/item/1005004751205589.html?spm=a2g0o.order_list.order_list_main.17.51041c24fInrw3&gatewayAdapt=](url)
to test new plugins (load max_ESP32_16M8M ) unfortunately does not work.
The terminal gives a message: _invalid header: 0xffffffff_

TD-er, 2023-04-15:
Yep, is for sure planned, along with the ESP32-C3 series.
I even have quite a few test units on my desk, but have not yet started due to lack of time for some extended amount of time as I know I need to do this in quite some longer runs.
This is not something that can be done in some spare hours as you need to do stuff in quite a lot of places. If done in 'lost hours', it will take forever.

TD-er, 2023-05-02:
Just started implementing support for ESP32-S3 and ESP32-C3: https://github.com/letscontrolit/ESPEasy/pull/4640

See first GH Actions build: https://github.com/letscontrolit/ESPEasy/actions/runs/4855900901

To my own surprise, it builds and on the ESP32-C3 it also boots and can connect to WiFi.
GPIO pin selection is not yet done and free memory detection is still incorrect.

ESP32-S3 I could not yet find to test with.
But to be sure I can test quickly with ESP32-S3, I also ordered a few modules from a Dutch web shop, so I hopefully have those by Wednesday for testing.

No idea what the current code will do on an ESP32-S3 as those seem to have only USB for serial, so test at your own rist as it might be tricky to flash a new build on it if this one fails.

PR #4640 (2023-05-10): [ESP32-C3 / -S3] Add support for ESP32-C3 & ESP32-S3
#fixes: #4614
#fixes: #3863

ToDo ESP32-C3:
- [x] Implement pin definitions
- [x] Check settings struct for boot states
- [x] Implement hardware specifics (e.g. USB behavior, SPI busses)
- [x] Add proper defaults for SPI busses etc.
- [x] Fix determine free stack
- [x] Document
- [x] Test

ToDo ESP32-S3:
- [x] Implement pin definitions
- [x] Check settings struct for boot states
- [x] Implement hardware specifics (e.g. USB behavior, SPI busses)
- [x] Add proper defaults for SPI busses etc.
- [x] Document
- [x] Test

N.B. support for CDC (con [... cut]