letscontrolit/ESPEasy #4620: Longpulse command with repetition always ends with high

Sminter4, 2023-04-19:
I use the version "ESP_Easy_mega_20230306_display_ESP8266_4M1M Mar 6 2023".
To create a repeated sound I want to use the following command:
LongPulse_mS,<GPIO>,<state>,<duration high>,<duration low>,**<nr of repeats>**
For me: longpulse_ms,1,0,50,50,3

If I only send one pulse (longpulse_ms,1,0,50,50,0), the GPIO finally goes low again.
As soon as I send more than one repetition, the GPIO always ends up high and I have the buzzer on continuous beeping.

Is this a bug in the code or the operation?

TD-er, 2023-04-19:
And what if you switch the 'state' argument?

Sminter4, 2023-04-19:
Both commands end after 3 single beeps with a continuous tone:
longpulse_ms,1,0,50,50,3
longpulse_ms,1,1,50,50,3

TD-er, 2023-04-19:
What does the pin states page tell you about this pin after the longpulse command?

TD-er, 2023-04-19:
You were using GPIO-1 so that's the 2nd line.
That is set to "output" and state 0 according to the pin states.
So I don't see how it is apparently actually "high" according to your beeper.

Sminter4, 2023-04-19:
My buzzer is low-active:
0=noise
1=mute

So he always ends up humming. :-(

Sminter4, 2023-04-19:
> And what if you switch the 'state' argument?
> 
> Edit: I notice you're using GPIO-1. That one is often wired to an USB to serial chip. Perhaps that one pulls the pin hight and ESPEasy may perhaps set the pin to input mode after the longpulse (which is still not desired behavior)

I have now also tried the command with a different GPIO (D8/GPIO15). The behavior is the same. Both commands end after 3 single beeps with a continuous tone (0=noise). 
```
longpulse_ms,1,0,50,50,3
longpulse_ms,1,1,50,50,3
```

TD-er, 2023-04-19:
GPIO-15 does have a pull-down resistor on an ESP8266 :)
See: https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp8266

Sminter4, 2023-04-19:
Thanks for the hint. I only connected the buzzer after the boot process... (Unfortunately I only have these two GPIOs free in my project.)

What is interesting is that with GPIO15 and GPIO1 the command works fine without repeating:
`longpulse_ms,1,0,50,50,0`

I only get a continuous beep (0=noise) if the number of repetitions is greater than zero.

tonhuisman, 2023-04-19:
@Sminter4 I have created a PR to fix this issue, please test using the download available from [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4747712196), once that's finished, and report your findings here.

Sminter4, 2023-04-20:
Tested with your version: "ESP_Easy_mega_20230419_display_ESP8266_4M1M Apr 19 2023"
and the GPIO1.
Unfortunately I could not find any difference to the previous version. Both commands end with a continuous pips:

```
longpulse_ms,1,0,50,50,3
longpulse_ms,1,1,50,50,3
```

However, this command still works correctly. With a single beep.
`longpulse_ms,1,0,100,100,0`

TD-er, 2023-04-20:
On ESP8266 we use a SDK call to generate the pulses in the background as this is way more stable compared to the ESPEasy scheduler.
But apparently the end state of this is always a 'full' period of up- and down-level.
So we need to add a scheduled call to turn the level  'up' on "inverted" state for longpulse calls with a (fixed) number of pulses.

tonhuisman, 2023-04-20:
A new build is in [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4758203319), where I reset the GPIO pin to the required state after the longpulse_ms sequence is completed.

Using a repeat of 0 was not involved in this issue, it only affected the _repeating_ longpulse(_ms) command where the main pulse is the low (0) value, and only on ESP8266.

Sminter4, 2023-04-21:
Successfully tested with the version "ESP_Easy_mega_20230420_display_ESP8266_4M1M Apr 20 2023" and the GPIO1:

Now this command also ends with a "1/high" as expected:
longpulse_ms,1,0,50,50,3

Thanks for your time and the quick fix!
This is how projects are fun!

[image]

uzi18, 2023-04-21:
@Sminter4 what is propose for right knob?

Sminter4, 2023-04-21:
> @Sminter4Was wird für den rechten Knopf vorgeschlagen?

There is a brightness sensor at the top, a rotary switch on the right, and the said buzzer at the bottom.
the Rotary Switch can be used to scroll through the menu. Time, date, power consumption, cistern level, children's account balance...

closed 2023-04-21 by Sminter4

tonhuisman, 2023-04-21:
Usually, when a PR is linked to an issue, the issue will be closed once the PR is merged. Now someone could get the impression that this fix is already merged into `mega` which it isn't yet.
No need to reopen this one, just a reminder for a next issue 😃

PR #4621 (2023-04-21): [LongPulse_ms] Fix end-state for inverted repeated pulse
Resolves #4620 

Bugfix (only for ESP8266):
- When having an inverted, repeated, pulse sequence, ensure the duration is long enough to return the pin to the initial (high) state (see linked issue for an example)

TODO:
- [x] Testing by reporter ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/4620#issuecomment-1517329073))