letscontrolit/ESPEasy #4622: MQTT - ESPEasy publish messages without the unit number if unit numer is 0

iz8mbw, 2023-04-20:
Hello.
ESPEasy publish MQTT messages without unit number if unit number is 0.
How is ESPEasy and MQTT confgured:
[image]

[image]

[image]

As you can see from configurations, the unit  umber is `espeasy_0` butt he MQTT messages doesn't report the `%unit%` ("0"):

```
espeasy/bme/Temperature 19.52
espeasy/bme/Humidity 62.70
espeasy/bme/Pressure 1016.28
espeasy/bme/Gas 27929
```

Instead, if the unit number is 1, the MQTT messages have also the unit number (__UnitNumber_):

```
espeasy_1/sht/Temperature 19.30
espeasy_1/sht/Humidity 63.08
```

TD-er, 2023-04-20:
A unit number of 0 is in ESPEasy somewhat of a "default-but-should-not-be-used" number.
Originally the unit nr was only used for the p2p layer, but later on it is being used in several other places.

In ESPEasy p2p messages the unit nr of 0 cannot be used to send commands as there is absolutely no guarantee there is an unique unit nr in your network. (with others this is also not guaranteed, but at least the user has set some value)
Another special value is 255 as that's internally used as a "broadcast ID".

So unit id 0 should simply not be used.

iz8mbw, 2023-04-23:
Ok, so to recap when we have more than one ESPEasy unit the lower value to assign is 1 and not zero.
Thanks.

closed 2023-04-23 by iz8mbw

tonhuisman, 2023-04-23:
Just be sure to assign each ESPEasy unit a unique unit number in the range 1..254