letscontrolit/ESPEasy #4627: Add Plugin 102 - PZEM004T into MAX build

iz8mbw, 2023-04-26:
Please can you add the Plugin 102 - PZEM004T into MAX build?
I see there is enough space into the 16MB bin.

Thanks.

TD-er, 2023-04-26:
Hmm that's strange, all plugins should be in the MAX builds (all except those not intended for ESP32...)

iz8mbw, 2023-04-26:
Yes sorry, I found it:
[image]

but as you can see its name doesn't start with "ENERGY" like the others:
[image]

iz8mbw, 2023-04-26:
indeed the "PZEM-004Tv30-Muliple" looks like the only one Plugin without a "classification" that generally is used (as text description) before the plugin name.

TD-er, 2023-04-26:
Ah so that should be changed :)

iz8mbw, 2023-04-26:
I suppose. A standard is a standard. :-)

TD-er, 2023-04-26:
> I suppose. A standard is a standard. :-)

[Yep](https://xkcd.com/927/) :)

iz8mbw, 2023-04-26:
:-)

PR #4628 (2023-04-27): [P102] Bring plugin-name in line with other plugins
Resolves #4627 

Features:
- Adjust plugin-name to the standard naming scheme `Category - plugin name` (and align with the documentation)
closed 2023-04-27 by TD-er