letscontrolit/ESPEasy #4633: Suggestion for improvement to "P104_max7219_Dotmatrix.ino": Dots can be controlled individually

Sminter4, 2023-04-29:
Hello everyone, I hope this is the right place for a suggestion for improvement.
I am currently working a lot with the dot matrix and would like to build a robot with "emotions" with it. Here you can find an example:
[image]

[https://www.hackster.io/igorF2/controlling-an-led-matrix-with-arduino-uno-0a9e94](https://www.hackster.io/igorF2/controlling-an-led-matrix-with-arduino-uno-0a9e94)

Of course, in the beauty of the ESPEasy. :-) Is it possible to control the dot matrix individually via ESPEasy?

tonhuisman, 2023-04-29:
I like the idea, and have thought about separately addressable pixels when developing the plugin, but didn't implement it then because of the added code-size of this already quite big plugin.
I'll put this feature request on my TODO list 👍

Sminter4, 2023-04-29:
That sounds good!
I'm already looking forward to the next mail trigger of this conversation.

tonhuisman, 2023-08-13:
@Sminter4 I've (finally) found some time (and inspiration) to add a feature to draw individual dots on a display-zone, please try [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/5849249097) result once it's ready, and report your findings for further improvement 😃

Sminter4, 2023-08-16:
Hey great that you thought of it! 👍
I have uploaded and tested the SW. So far everything is going as expected. Thanks!
[image]

TD-er, 2023-08-16:
Decoding that will probably add quite a lot to the bin size.
Also animated gif can be quite hard to process as you can have an arbitrary "framesize" per frame (as long as it is a rectangle), so you must keep track of the previous frame before rendering the next one.

uzi18, 2023-08-17:
@TD-er it is very small resolution

tonhuisman, 2023-08-17:
AFAICS, gif or png are compressed by default, having these uncompressed would seem _very_ weird. And while I can think of ways this could be useful, the anticipated code-size increase (2 separate decoders) would make it impractical to have it available in any build except the MAX builds. Also, it would seem more usable on a TFT display. And the repeating nature of f.e. an animated gif will open a whole new box of Pandorra...
I'll do some investigation.

closed 2023-09-17 by TD-er

Sminter4, 2023-10-07:
One more question. Is there a tool that I can use to calculate the code for a graphic?
E.g. based on an 8x8 bmp-graphic-file.

I think ChatGPT could do it, but I obviously haven't found the right prompt yet.
("Can you write me a bird with the following code: DotMatrix,Dot,<zone>,<r>,<c>[,0][,...]
r = Row coordinate, range 1..8.
c = Column coordinate, range 1..8 x module count.")

tonhuisman, 2023-10-07:
Haven't used ChatGPT for anything useful yet, so I have no clue on how to prompt for that 😆 

As an alternative, I could add some features to the `dot` subcommand, like switching between bit/dot and byte/hex mode, so you can write an entire column (8 dots/bits) with less data, f.e.
`dotmatrix,dot,<zone>,<r>,<c>,x,<byte>[,<byte>,...],b,<r>,<c>[,0],...`
where the `x` switches to byte/hex mode, that auto-increments the column for each byte provided, and `b` switches to the default bit/dot mode, without auto-increment.

Sminter4, 2023-10-08:
Unfortunately, I couldn't fully understand your alternative option. But I suspect that with different spellings there will be a simpler one...
If programming an alternative spelling isn't a problem, I'd be happy to try it out.

tonhuisman, 2023-10-08:
> I couldn't fully understand your alternative option

I'm trying to suggest a way to not have to address each separate dot, but 1 column at a time, possibly making the drawing of a larger figure a bit simpler/compacter. But this would be more technically oriented, making it maybe even _less_ simple 😨. Let's keep it as it is for now, and see how this works out once you have trained ChatGPT properly 😆

PR #4762 (2023-09-17): [P104] Add Dot subcommand to draw individual dots in a zone
Resolves #4633 

Features:
- Add `DotMatrix,Dot,<zone>,<r>,<c>[,0][,<r>,<c>[,0]...]` subcommand, to draw individual dots in any zone
- Update documentation
- Other code improvements

TODO:
- [x] Testing by requester ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/4633#issuecomment-1680948198))