letscontrolit/ESPEasy #4635: Oled Framed font settings non functional since about Release mega-20230306 possibly earlier.

Budman1758, 2023-04-30:
Not sure what release this showed up with but its pretty recent. OLED Framed plugin will not save or use different font settings at the line by line settings part of the page. No matter what size setting I try it always reverts back to all 12 lines @ "use smallest" and the display does not reflect that setting.  On my 3 units this is happening on I always use the "reduce to smaller" setting and this is what that looks like....

[image]

Now what I get is this.  Notice how the line does not fit.

[image]

During my testing for this I noticed that when I went to a newer build with the problem at first the display was correct. It changed AFTER I went to the plugin settings and made a minor change.  This is using the Display bin files. This issue is true for both ESP32 and ESP-8266 as I have 3 units doing this. 2 are ESP32. All are SSD1306 128x64 2.4 inch I2C displays.

Let me know what info you need to help.  Thanks.

tonhuisman, 2023-04-30:
Have you tried reverting to a somewhat older release, to determine a time-frame this might have been introduced?

Budman1758, 2023-04-30:
Revert to Release mega-20221224 and it works as it should.

tonhuisman, 2023-04-30:
So if you correct the settings using that older build, it will run correctly on newer builds _until_ you change (Submit) your display's settings page?

tonhuisman, 2023-04-30:
Update: I think I've found it, will create a PR soon.

tonhuisman, 2023-04-30:
@Budman1758 A build is cooking in [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4845586058), please report your findings here or in the PR 😃 (you know the drill 😛).

Budman1758, 2023-04-30:
> @Budman1758 A build is cooking in [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4845586058), please report your findings here or in the PR 😃 (you know the drill 😛).

Will do. Gotta hit a grankids birthday party first. :>)

Budman1758, 2023-05-01:
It would appear that you have squashed that particular bug. All 3 units updated and working as they should as far as I can tell.

PR #4636 (2023-05-01): [P036] Bugfix for overlapping page variables
Resolves #4635 

Bugfix:
- The per line Font change and Alignment change used the same page-variable.

TODO:
- [x] Testing by issue-reporter ([Confirmed](https://github.com/letscontrolit/ESPEasy/issues/4635#issuecomment-1529207552))
closed 2023-05-01 by TD-er