letscontrolit/ESPEasy #4646: Input - iButton state is not send to mqtt

khoebling, 2023-05-04:
Hi!

The iButton Value is not send to the MQTT Broker. I tried also with Switch Input -Input an this value is transmitted to the MQTT Broker.

The log file shows that the button is detected, the value on the web interface changes from 0 to 1 and vice versa

Karl

TD-er, 2023-05-04:
Probably there needs to be a call to `SendData` when the event is also generated.
Or a scheduled call to `PLUGIN_READ` when it is read.

tonhuisman, 2023-05-04:
I've added state-change based updates, on the scheduled interval, to controllers.

khoebling, 2023-05-04:
<html>
<body>

Git Build: | HEAD_9acaaf8
-- | --

</body>
</html>

Sends to MQTT Broker but value is 65537 - shouldnt it be 1

TD-er, 2023-05-04:
> Git Build:	HEAD_9acaaf8
> Sends to MQTT Broker but value is 65537 - shouldnt it be 1

That's because the next userVar value is being used to store the previous value.
But for type long the values on index 0 and 1 are being combined.

Better store the type in one of those recently added output types.
Also how many bits are needed to send an iButton value? Does it fit in a 32 bit int?

closed 2023-05-05 by TD-er