letscontrolit/ESPEasy #4649: ESP_Easy_mega_20230505_normal_ESP8266_4M1M don't receive ESPEasy P2P Networking values

fly74, 2023-05-06:
After update from last official to ESP_Easy_mega_20230505_normal_ESP8266_4M1M the device don't receive P2P values. The ESP32 devices with ESP_Easy_mega_20230505_normal_ESP32_4M316k get the values. Looks like a ESP8266 related issue.

tonhuisman, 2023-05-06:
Have you tried to press the reset button or power-cycle the unit, after the update?

fly74, 2023-05-06:
Yes, after power-cycle the esp32 it get no values too. It seem s 20230505 P2P don't work. Back to 20230409 works.

tonhuisman, 2023-05-06:
> after power-cycle the esp32

I was thinking of a reset/power cycle of the ESP8266 unit... 🤔

fly74, 2023-05-06:
I power-cycled both to be clear. :/

TD-er, 2023-05-06:
I will have a look at it.
Just to be sure, you are talking about receiving values from a task on another node?

fly74, 2023-05-06:
Yes, I'm using the "ESPEasy P2P Networking"-controller for sending from a BME280 Task to the network.

fly74, 2023-05-06:
@TD-er What means the release note: "[TaskValueTypes] Simplify ESPEasy p2p task value exchange"

TD-er, 2023-05-06:
Well that's probably the commit where I messed up ;(
I have introduced a few new "Dummy Types" to store, like 64 bit ints, upto 4 int32 or uint32.
So I also had to make sure those new types could be shared among p2p nodes.
Thus I removed some code duplication where there were a number of tests in the ESPeasy code to handle the existing "LONG" type (which was essentially an unsigned type) or upto 4 task values.
But I guess this may have introduced the bug you're seeing now.

My first guess is, it might be just an alignment issue, so instead of just reading/writing the byte array at the position where this struct is, I may need to memcpy the bytes before processing them.
a 64 bit value may have other alignment requirements and I think this is also where Windows and Linux compiles may yield different results.
So I will also add some compile-time checks to make sure these values in the p2p data struct remain at the same offset.

TD-er, 2023-05-06:
Made a quick fix (not yet tested): https://github.com/letscontrolit/ESPEasy/pull/4650

When the test build is ready, are you able to test it?  https://github.com/letscontrolit/ESPEasy/actions/runs/4901603897

fly74, 2023-05-06:
Sure.

TD-er, 2023-05-06:
Hmm I am also testing it on a ESP32-C3 unit with this fixed code and try to share values from another node running a build of about a month ago.
But it doesn't seem like it is working well.... So I would love to see your conclusion to make sure it is not a bug in my setup here :)

fly74, 2023-05-06:
> When the test build is ready, are you able to test it? https://github.com/letscontrolit/ESPEasy/actions/runs/4901603897

@TD-er No change, the values are not received.

Sensor ESP is running: ESP_Easy_mega_20230506_normal_ESP8266_4M1M
Receiving ESP running: ESP_Easy_mega_20230506_normal_ESP8266_4M1M and ESP_Easy_mega_20230506_normal_ESP32_4M316k

fly74, 2023-05-06:
> So I would love to see your conclusion to make sure it is not a bug in my setup here :)

@TD-er When I flash back to 20230409, it works instantly.

TD-er, 2023-05-06:
And do you need to revert the sending or the receiving node to the previous build to get it to work again?

fly74, 2023-05-06:
The only thing is, that I shiftet the UDP port für the P2P controller in the advanced setting. The tasks have only the option send to controller with it's intervall.

fly74, 2023-05-06:
> And do you need to revert the sending or the receiving node to the previous build to get it to work again?

No, flash back to 20230409 works after flashing. Without any changes in setting.

TD-er, 2023-05-06:
OK, found the bug...
There was some extra check added to make sure the sending node uses the same plugin ID.
But apparently it may not have been set on the sending node (at least not the node here running an older build)

And while testing it I also found some other issue.
The receiving task doesn't trigger all kinds of actions which would have happened on a normal call to PLUGIN_READ.
For example the plugin statistics are not updated and probably it will also not send to any linked controller.
So I will also work on that.

I will have a fix later this evening for all these, but first I will make a commit for you to test with these changes....

TD-er, 2023-05-06:
Added support for sending the p2p received data to any connected controller.
Also the stats now work on p2p received data.

Not sure what to do with sending to the p2p controller when receiving data from a p2p source.

One other thing that's a bit 'tedious' is that the task config is not being copied when initializing a remote fed task.
For example sharing a sysinfo task will not set the output units of measure.

But that never has worked, so that must be a new feature to add :)

TD-er, 2023-05-06:
OK, apparently the latest test build failed.
So fixed that part and it should be present in this test build:
https://github.com/letscontrolit/ESPEasy/actions/runs/4903091778
Please let me know if this does fix your p2p issues.

fly74, 2023-05-06:
> OK, apparently the latest test build failed. So fixed that part and it should be present in this test build: https://github.com/letscontrolit/ESPEasy/actions/runs/4903091778 Please let me know if this does fix your p2p issues.

It's a little strange this release can receive values from nodes with the official 20230505 but not from nodes with this action build ?!?

TD-er, 2023-05-06:
huh?
That's really strange, will look into it

TD-er, 2023-05-06:
Ah found something that's no longer really strange, but more like a stupid copy-paste error of mine....

Can you please check this build: https://github.com/letscontrolit/ESPEasy/actions/runs/4903565952

fly74, 2023-05-07:
> Can you please check this build: https://github.com/letscontrolit/ESPEasy/actions/runs/4903565952

@TD-er Yeah, that is working!

TD-er, 2023-05-07:
Thanks for testing

closed 2023-05-07 by TD-er

TD-er, 2023-05-07:
@fly74 The new release has been over 10 minutes already and no new reported bug yet? ;)

fly74, 2023-05-07:
@TD-er Now I'm back give me a second.

fly74, 2023-05-07:
Rollout started. Wait..... Reboot...., no issues looks fine!

TD-er, 2023-05-07:
Great! :)
Thanks.

TD-er, 2023-06-22:
Since you always manage to find bugs in no-time, could you please test the latest code merges?
See: https://td-er.nl/ESPEasy/latest/
I used the build files from this GH Actions build: https://github.com/letscontrolit/ESPEasy/actions/runs/5343779493

fly74, 2023-06-22:
@TD-er I rolled out the follwing:

```
ESP_Easy_mega_20230622_climate_ESP8266_4M1M
ESP_Easy_mega_20230622_neopixel_ESP32_4M316k
ESP_Easy_mega_20230622_normal_ESP32_4M316k
ESP_Easy_mega_20230622_normal_ESP8266_4M1M
```

Tomorrow I give you a feedback. What's the weather in Nederland, Germany is sweating.

TD-er, 2023-06-22:
It is OK here in the north. Not too warm, not that humid.
In the south there is going to be a lot of rain and some of it may end up in Germany as I understood.

Thanks for testing :)
When nothing serious is found, I probably will make a new build this weekend.

fly74, 2023-06-22:
@TD-er Nothing serious found, everything of my landscape is working!

TD-er, 2023-06-22:
> @TD-er Nothing serious found, everything of my landscape is working!

Wow, we should have tried this before :)
This is especially impressive as there have been a lot of changes since the last build which could impact a lot of the system.

fly74, 2023-06-22:
Then let it from the chain.

TD-er, 2023-06-22:
Well Ton also wanted to test a few things.

fly74, 2023-06-22:
@tonhuisman Hurry up to release the beast.

TD-er, 2023-06-22:
Well he did find a bug....
Fix here: https://github.com/letscontrolit/ESPEasy/actions/runs/5350372132
And the web flasher also has been updated with this build: https://td-er.nl/ESPEasy/latest/

fly74, 2023-06-23:
> Well he did find a bug.... Fix here: https://github.com/letscontrolit/ESPEasy/actions/runs/5350372132 And the web flasher also has been updated with this build: https://td-er.nl/ESPEasy/latest/

@TD-er No problems with this one.

PR #4650 (2023-05-07): [p2p] Fix task value data exchange between nodes via p2p (#4649)
Fixes: #4649
commit 346467a3 (2023-05-07): [p2p] Fix task value data exchange between nodes via p2p (#4649)

Fixes: #4649