letscontrolit/ESPEasy #4652: UserVar index out of range

chunter1, 2023-05-07:
I just compiled the latest Release mega-20230507 and now i am getting the error "UserVar index out of range".
Did i miss something or is it a bug?

tonhuisman, 2023-05-07:
What build are you compiling?
And can you show some more details of that message?

chunter1, 2023-05-07:
I got "default_envs = normal_ESP8266_4M1M" set.
For testing i configured a "Generic - System Info" device with "uptime" as "Value 1" set.
In the Log i get the messages "UserVar index out of range" ech time the device is triggered.

tonhuisman, 2023-05-07:
Did you also test using the official downloadable files? As during replay of your scenario (using an ESP32-S2 that's currently connected), and having the Sysinfo plugin on both Task 1 and Task 32, I don't see that message in the (serial) log.
I'll retest using an ESP8266 in a bit.

tonhuisman, 2023-05-07:
I'm on to something 🤔, working on a fix.

tonhuisman, 2023-05-07:
@chunter1 A build is running in [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4908743943), can you please test when that's finished and report your findings here (or in the PR, linked above this message)?

chunter1, 2023-05-08:
The version you sent me does not have that issue and works as expected.

TD-er, 2023-05-08:
Great!
Then I will merge it and make a new build for it, as it is a truly bad user experience to be unable to use ESP8266 with yesterday's build.

PR #4653 (2023-05-08): [UserVar] Bugfix: Clear element only
Resolves #4652 

Bugfix:
- UserVar data element should be cleared, not entire list. (Only found to fail on ESP8266)

TODO:
- [x] Testing by reporter
closed 2023-05-08 by TD-er