letscontrolit/ESPEasy #4655: ESP8266: no fallback to AP mode if STA connection fails

chemmex, 2023-05-08:
Sorry for raising this again, but it seems that latest ESP8266 versions do not fall back to the AP mode if STA connection fails. I tried any condition: configured AP not available, wrong wifi credentials. The only case it is started is if no SSID configured. 
Though, I have several units running 2020 code, they handle it correctly.

There is a long discussion in #4337, but it is still unclear if the fallback to AP is forcibly disabled or this is a try and fail approach.

TD-er, 2023-05-08:
Do you have "Do Not Start AP" checked on the config page?

chemmex, 2023-05-08:
No

dpancer, 2023-06-17:
I"m having the same issue. I have a nodemcu and have flashed it with Normal / Climate / Collection A. All firmwares do not fall back to AP Mode.

burdal1, 2023-07-25:
I have the same problem. If the any SSID is configured (and this SSID is not available at the moment), the AP mode don't start. If SSID isn't configured, the AP starts correctly.

ESP_Easy_mega_20230623_energy_ESP8266_4M1M

jelinekpetr, 2023-09-06:
I have the same issue in 20230822 for ESP8266... Only way to start AP is no SSID is filled up. Any SSID (default one, deployment or support ones) filled up causes AP mode doesn't start even if the SSIDs are out of reach.

tonhuisman, 2023-11-11:
@chemmex @dpancer @burdal1 @jelinekpetr I've found and fixed an issue that prevented AP mode from starting when no matching AP was found, you can download form [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/6836421268) for testing, or pull PR #4875 to confirm that fix resolves the issue.