letscontrolit/ESPEasy #4659: Open the latest version(mega-20230508 ESP8266) setting page is much slower than old version(e.g. mega-20220427  ESP8266)

eznas, 2023-05-11:
I played with ESPEasy about 1 year ago, I love its control style, rules, web pages...

I download the latest version and flashed into my ESP8266 NodeMCU board to find whatnew for this project. But I soon found it takes much more time to load and switch the setting pages. The lateast version takes 20-30 seconds to load a new page, but the old one takes less than 5 seconds! 

I doubted the flash chip speed or quality may be the cause, so I tried flash old version firmware into same board. 
Then the speed cames back!

I am sure the newer version supports more devices or functions, but the setting pages experience is auwful...

### Summarize of the problem/feature request

Slower pages switching.

### Expected behavior

Speed up the pages loading.

### Steps to reproduce

1. please download and flash the old version (e.g. 20220424) as comparision
2. Switch setting pages and count the time

TD-er, 2023-05-11:
Which exact bin file are you using?

The page load here is quite OK.
To be honest, even "seconds" of load time is way too much.
It should be upto several 100s of msec.

So I guess it might be related to loading the CSS via the CDN servers.
On last Tuesday and Wednesday the servers of GitHub were having lots and lots of issues.
That's why I think the cached versions of the JS and CSS files may thus not load from those servers.

This morning I tried to download a GH-Actions build (200+ MB) and it downloaded at a speed of 600-ish kB/sec.

eznas, 2023-05-12:
I finally found the problem! 

When I look into the webpage source code and I found the page will load ""https://ajax.googleapis.com/ajax/libs/jquery/3.6.0/jquery.min.js", but  this site was filter out by GFW(The Great Fire Wall of China)，so very time I reload the setting page in China, the browser will tried to load this js file till the time out...

As my suggestion, use "https://code.jquery.com/jquery-3.6.0.min.js" or "https://cdn.bootcss.com/jquery/3.6.0/jquery.min.js" instead for the page code. 

Then it will be much faster loading speed in China.

TD-er, 2023-05-12:
Sure will do!

Thanks for looking into it.