letscontrolit/ESPEasy #4662: WIFI : No valid wifi settings ESP8266 Wemos D1 mini

Jumbo125, 2023-05-12:
Hello
i want to flash my wemos mini d1 with the ESP_Easy_mega_20230508_collection_A_ESP8266_4M1M

- i add my Router in the WLAN settings (Wifi main and fallback)
- i enable post-flash action 
- i use baudrate 115200
- after flashing i disconnect the wemos and reconnect it to my pc and show the serial output.

logs looks fine:
```
[11.05.2023 23:03:35] 	erasing flash
[11.05.2023 23:03:35] 	size: 0fdc90 address: 000000
[11.05.2023 23:03:35] 	first_sector_index: 0
[11.05.2023 23:03:35] 	total_sector_count: 254
[11.05.2023 23:03:35] 	head_sector_count: 16
[11.05.2023 23:03:35] 	adjusted_sector_count: 238
[11.05.2023 23:03:35] 	erase_size: 0ee000
[11.05.2023 23:03:35] 	espcomm_send_command: sending command header
[11.05.2023 23:03:35] 	espcomm_send_command: sending command payload
[11.05.2023 23:03:35] 	setting serial port timeouts to 15000 ms
[11.05.2023 23:03:35] 	setting serial port timeouts to 1000 ms
[11.05.2023 23:03:35] 	espcomm_send_command: receiving 2 bytes of data
[11.05.2023 23:03:35] 	writing flash
[11.05.2023 23:03:43] ................................................................................ [  7% ]
[11.05.2023 23:03:52] ................................................................................ [ 15% ]
[11.05.2023 23:04:00] ................................................................................ [ 23% ]
[11.05.2023 23:04:09] ................................................................................ [ 31% ]
[11.05.2023 23:04:17] ................................................................................ [ 39% ]
[11.05.2023 23:04:25] ................................................................................ [ 47% ]
[11.05.2023 23:04:34] ................................................................................ [ 55% ]
[11.05.2023 23:04:42] ................................................................................ [ 62% ]
[11.05.2023 23:04:50] ................................................................................ [ 70% ]
[11.05.2023 23:04:59] ................................................................................ [ 78% ]
[11.05.2023 23:05:07] ................................................................................ [ 86% ]
[11.05.2023 23:05:15] ................................................................................ [ 94% ]
[11.05.2023 23:05:21] ........................................................                         [ 100% ]
[11.05.2023 23:05:21] starting app without reboot
[11.05.2023 23:05:21] 	espcomm_send_command: sending command header
[11.05.2023 23:05:21] 	espcomm_send_command: sending command payload
[11.05.2023 23:05:21] 	espcomm_send_command: receiving 2 bytes of data
[11.05.2023 23:05:21] closing bootloader
[11.05.2023 23:05:21] 	flush start
[11.05.2023 23:05:21] 	setting serial port timeouts to 1 ms
[11.05.2023 23:05:21] 	setting serial port timeouts to 1000 ms
[11.05.2023 23:05:21] 	flush complete
[2023-05-11 23:05:21] DONE!
######POST FLASH######
############################
[2023-05-11 23:05:23] JOB COMPLETED, over and out!
```
i always get the same error

```
5220 : Info   : Setup: Scan all channels
7403 : Info   : INIT : Free RAM:27680
7624 : Info   : INFO : Plugins: 80 [Normal][Collection][Collection_A][No Debug Log] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
7628 : Info   : Webserver: start
8970 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 21032 WiFiStatus 0 ESPeasy internal wifi status: DISCONNECTED
10226 : Error  : WIFI : No valid wifi settings
14599 : Info   : WiFi : WiFiConnected(), start AP
16881 : Info   : WIFI : Set WiFi to AP+STA
17873 : Info   : WIFI : AP Mode ssid will be Collection-A with address 192.168.4.1
```

any solution?
thank you

tonhuisman, 2023-05-12:
Writing the WiFi settings from the ESPEase.Flasher doesn't seem to work.
Alternative:
- Connect to the serial port using a terminal application (not just a viewer), baudrate: 115200, No parity, 8 databits, 1 stopbit
- Type these commands: (complete each line by pressing the Enter key)
  `wifissid,<your_wifi_ssid>`
  `wifikey,<your_wifi_password>`
  `save`
- NB: While typing these commands there will be no echo, and no way to correct typing errors!
- If after the change of the WiFiKey, no reconnect is triggered, you can type `wifidisconnect`, then the unit will reconnect to wifi
- If you made a typo in the ssid or password, just retry that command with the correct data
- Commands are not case-sensitive, but passwords are!

Jumbo125, 2023-05-15:
thank you for your answer, but i pass it with normal arduino ide

TD-er, 2023-05-15:
Just any serial terminal program will do.
Baud rate is 115200.

tonhuisman, 2023-09-09:
@Jumbo125 assuming this is solved by now as there has been no more response, I guess we can close the issue?

TD-er, 2023-09-09:
Close for now, whenever it is still an issue, let us know.

closed 2023-09-09 by TD-er