letscontrolit/ESPEasy #4667: Sudden reboot after setting GPIO 5

DrBakterius, 2023-05-14:
If I set GPIO 5 (Switch input - Switch) to '1', the ESP reboots. But this does not happen every time but very often. Build: ESP_Easy_mega_20230409_normal_ESP8266_4M1M Apr 9 2023 (but also earlier ones). Disconnecting the power does not solve the problem.

[image]

Log:
```
2023-05-14T09:38:04.465729+02:00 Aquarium-2 EspEasy: GPIO,5,0
2023-05-14T09:38:04.468177+02:00 Aquarium-2 EspEasy: Par1: 5 Par2: 0 Par3: 0 Par4: 0 Par5: 0
2023-05-14T09:38:04.470915+02:00 Aquarium-2 EspEasy: GPIO : port#5: set to 0
2023-05-14T09:38:04.557919+02:00 Aquarium-2 EspEasy: SW  : GPIO=5 State=0 Output value=1
2023-05-14T09:38:04.574404+02:00 Aquarium-2 EspEasy: MQTT : Aquarium/Skimmer/Status 1
2023-05-14T09:38:04.588431+02:00 Aquarium-2 EspEasy: EVENT: Skimmer#Status=1
2023-05-14T09:38:04.595845+02:00 Aquarium-2 EspEasy: EVENT: Skimmer#Status=1 Processing time:8 milliSeconds
2023-05-14T09:38:05.741986+02:00 Aquarium-2 EspEasy: Command: GPIO
2023-05-14T09:38:05.744579+02:00 Aquarium-2 EspEasy: GPIO,5,1
2023-05-14T09:38:05.746795+02:00 Aquarium-2 EspEasy: Par1: 5 Par2: 1 Par3: 0 Par4: 0 Par5: 0
2023-05-14T09:38:05.749060+02:00 Aquarium-2 EspEasy: GPIO : port#5: set to 1
2023-05-14T09:38:08.788810+02:00 Aquarium-2 EspEasy: firstLoopConnectionsEstablished
2023-05-14T09:38:08.789977+02:00 Aquarium-2 EspEasy: SW   : State 0
2023-05-14T09:38:08.823682+02:00 Aquarium-2 EspEasy: SW   : State 1
2023-05-14T09:38:08.847955+02:00 Aquarium-2 EspEasy: SW   : State 1
2023-05-14T09:38:08.874791+02:00 Aquarium-2 EspEasy: SW   : State 1
2023-05-14T09:38:08.898846+02:00 Aquarium-2 EspEasy: Dummy: value 1: 1
2023-05-14T09:38:08.925602+02:00 Aquarium-2 EspEasy: EVENT: TaskInit#WLAN_Empfang=1,1
2023-05-14T09:38:08.940329+02:00 Aquarium-2 EspEasy: SW  : GPIO=12 State=1 Output value=1
2023-05-14T09:38:08.964406+02:00 Aquarium-2 EspEasy: SW  : GPIO=13 State=1 Output value=1
2023-05-14T09:38:08.992410+02:00 Aquarium-2 EspEasy: SW  : GPIO=5 State=1 Output value=0
2023-05-14T09:38:09.044223+02:00 Aquarium-2 EspEasy: MQTT : Intentional reconnect
2023-05-14T09:38:09.059020+02:00 Aquarium-2 EspEasy: MQTT : Connected to broker with client ID: Aquarium_2
2023-05-14T09:38:09.076465+02:00 Aquarium-2 EspEasy: Subscribed to: Aquarium/#
2023-05-14T09:38:09.089137+02:00 Aquarium-2 EspEasy: SW  : GPIO=4 State=1 Output value=0
2023-05-14T09:38:09.121044+02:00 Aquarium-2 EspEasy: EVENT: TaskInit#Wassertemperatur=2,1
2023-05-14T09:38:09.158970+02:00 Aquarium-2 EspEasy: EVENT: TaskInit#Heizung=3,1
2023-05-14T09:38:09.162943+02:00 Aquarium-2 EspEasy: EVENT: TaskInit#Pumpe=4,1
2023-05-14T09:38:09.170746+02:00 Aquarium-2 EspEasy: EVENT: TaskInit#Licht=5,1
2023-05-14T09:38:09.205637+02:00 Aquarium-2 EspEasy: EVENT: TaskInit#CO2=6,1
2023-05-14T09:38:09.205637+02:00 Aquarium-2 EspEasy: EVENT: TaskInit#Skimmer=7,1
2023-05-14T09:38:09.210236+02:00 Aquarium-2 EspEasy: EVENT: TaskInit#Luftpumpe=8,1
2023-05-14T09:38:09.225078+02:00 Aquarium-2 EspEasy: EVENT: TaskInit#Reboot=9,1
2023-05-14T09:38:09.246375+02:00 Aquarium-2 EspEasy: EVENT: WLAN_Empfang#Signal=-59
2023-05-14T09:38:09.255353+02:00 Aquarium-2 EspEasy: EVENT: WiFi#ChangedAccesspoint
2023-05-14T09:38:09.268882+02:00 Aquarium-2 EspEasy: EVENT: WiFi#ChangedWiFichannel
2023-05-14T09:38:09.276711+02:00 Aquarium-2 EspEasy: EVENT: WiFi#Connected
2023-05-14T09:38:09.290474+02:00 Aquarium-2 EspEasy: EVENT: Heizung#Status=1
2023-05-14T09:38:09.304654+02:00 Aquarium-2 EspEasy: EVENT: Pumpe#Status=0
2023-05-14T09:38:09.312171+02:00 Aquarium-2 EspEasy: EVENT: Licht#Status=0
2023-05-14T09:38:09.330298+02:00 Aquarium-2 EspEasy: EVENT: CO2#Status=1
2023-05-14T09:38:09.353464+02:00 Aquarium-2 EspEasy: EVENT: Skimmer#Status=1
2023-05-14T09:38:09.366769+02:00 Aquarium-2 EspEasy: EVENT: Luftpumpe#Status=1
2023-05-14T09:38:09.375680+02:00 Aquarium-2 EspEasy: EVENT: Reboot#Status=1
2023-05-14T09:38:09.387730+02:00 Aquarium-2 EspEasy: EVENT: Pumpe#Status=1
2023-05-14T09:38:09.402476+02:00 Aquarium-2 EspEasy: EVENT: Licht#Status=1
2023-05-14T09:38:09.412682+02:00 Aquarium-2 EspEasy: NTP  : NTP host 192.168.2.102 (192.168.2.102) queried
2023-05-14T09:38:09.431176+02:00 Aquarium-2 EspEasy: NTP  : NTP replied: delay 11 mSec Accuracy increased by 0.423 seconds
2023-05-14T09:38:09.434992+02:00 Aquarium-2 EspEasy: Time set to 1684049889.423
2023-05-14T09:38:09.438495+02:00 Aquarium-2 EspEasy: Current Time Zone:  DST time start: 2023-03-26 02:00:00 offset: 120 min STD time start: 2023-10-29 03:00:00 offset: 60 min
2023-05-14T09:38:09.440375+02:00 Aquarium-2 EspEasy: Local time: 2023-05-14 09:38:09
2023-05-14T09:38:09.451044+02:00 Aquarium-2 EspEasy: EVENT: Clock#Time=Sun,9:38
2023-05-14T09:38:09.457774+02:00 Aquarium-2 EspEasy: EVENT: Clock#Time=Sun,9:38 Processing time:7 milliSeconds
2023-05-14T09:38:09.472284+02:00 Aquarium-2 EspEasy: EVENT: Skimmer#Status=0
2023-05-14T09:38:09.479721+02:00 Aquarium-2 EspEasy: EVENT: Skimmer#Status=0 Processing time:7 milliSeconds
2023-05-14T09:38:09.495987+02:00 Aquarium-2 EspEasy: EVENT: MQTT#Connected
```

tonhuisman, 2023-05-14:
By default GPIO-5 is configured to be used by I2C (Hardware tab), is that still the case?
What is actually connected to that GPIO, a switch or something else?

DrBakterius, 2023-05-14:
Yes, GPIO 5 can be configured for the I2C (SCL) interface. However, this is not active for me.

[image]

I have a 5V relay module with optocoupler (low-level trigger) connected.

tonhuisman, 2023-05-14:
Can you check the power supply of the ESP?, as switching on or off a load (via a relay) can cause spikes or power-drops, and when these are large enough can cause a reset or crash of the ESP. This is especially important when a single power supply is used.
Separating the ESP power supply from other devices might be the solution here.

If needed the power supplies can have a shared ground, but first check if these grounds are potential-free, to avoid damage or dangerous situations! ⚠️

TD-er, 2023-05-14:
Also make sure to have a diode over the relay in the "blocking" orientation.

Problems may otherwise occur when switching a relay as is a coil.
A coil makes a magnetic field when current flows through it.
However, when disconnecting the relay's coil, the coil tries to keep the magnetic field in tact and thus generates a (much, much) higher negative voltage.
By adding a diode over the coil in the "reverse" direction, you immediately short this negative voltage before it can cause issues with the rest of the electronics connected to it.

Simple example:
[image]

Of course the power supply should also be sufficient, like Ton suggested.
And don't forget to use thick enough wires for the voltage supply nets.
Really flimsy breadboard wires often cause a significant voltage drop when the ESP draws its peak current, like with WiFi activity.

DrBakterius, 2023-05-14:
Many thanks for your advice and suggestions!

I use a mini power supply with 5V / 600mA. ([Picture](https://m.media-amazon.com/images/I/41hKB3q9IoL._SY445_SX342_QL70_ML2_.jpg)) I thought 3W would be enough. But maybe the peak current is too high.

The relay modules have a diode built in.  ([Picture](https://m.media-amazon.com/images/I/71Y+KDo75IL._SX522_.jpg)) So that can't be the problem.

TD-er, 2023-05-14:
600 mA is really tight as the ESP itself may draw upto 500 mA (really short spikes) when starting the WiFi radio.

Maybe you can add some extra capacitor on the ESP's 3V3 line?
Something like 220 or 330 uF between 3V3 and GND.
May need some experimentation as it also may 'boot issues'.
A discharged capacitor looks like a 'short circuit' as it will draw as much current as it can to charge.
So a too large capacitor may cause issues for the voltage regulator chip during boot.
Otherwise add this capacitor over the 5V and GND, but then it may not just help to buffer for the ESP, but just for the relay.

TD-er, 2023-05-15:
It was controller C005 (MQTT HomeAssistant/OpenHAB)
The controller function was: CPLUGIN_PROTOCOL_RECV

So it looks like the ESP received a command via MQTT.

A watchdog error means the ESP got 'stuck' executing some code.

Do you maybe send logs to a syslog server?
I don't see why the call to a GPIO may cause the system to lockup.
However if the voltage drops during such action, causing the WiFi to enter some undefined state, then any attempted communication via WiFi may cause a lockup.
UDP traffic is particularly susceptable to this and since the GPIO command code is adding some logs, I was wondering whether you might be sending your logs to a syslog server.

TD-er, 2023-05-15:
Can you disconnect GPIO-15 and try again this last setup? See [here](https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp8266) to understand why :)

Make sure you also connect GND between ESP and the REL board.
That's probably the most important line to connect as the voltages will otherwise be 'floating'.

DrBakterius, 2023-05-15:
Hm, why has it worked with GPIO 15 so far?

If you want galvanic isolation, GND must not be connected. I also prefer the separation with two power supplies. Here is a circuit diagram of how the relays are operated:

[image]
(from: [https://electronics.stackexchange.com/questions/505318/how-to-properly-use-a-relay-module-with-jd-vcc-from-arduino-raspberry](https://electronics.stackexchange.com/questions/505318/how-to-properly-use-a-relay-module-with-jd-vcc-from-arduino-raspberry))

And because it also reboots when I use two power supplies, I rebuilt everything back to 1. There are also no problems when booting.

By the way, in addition to GPIO 12-15, I also use GPIO 4 and 5 with such a relay module. A temperature sensor is connected to GPIO 2. According to my experience, the reboots only happen when the relay drops out, i.e. when the GPIO goes high. If the power supply was the problem, it should only happen when the relay is activated, right?

TD-er, 2023-05-15:
Please look at the table I linked.
On ESP8266, GPIO 0, 2 and 15 are so called 'strapping' pins, meaning the state of those pins during boot does matter.

GPIO-0 and 2 must be high during boot and GPIO-15 must be low during boot.

If not, then the ESP will not boot the ESPEasy sketch, but may enter flash mode.

If you want to use GPIO-15 as a normal output pin, you must use it via a PNP transistor.

When you're using optocouplers for isolation, then you should not have anything wired between the ESP and the REL board.
But for testing whether it will work, you can just use simple LEDs (+ some resistor of course) as the internals of the optocoupler are just LEDs.
This way you can test whether it may work using those LEDs and you can thus exclude the effects of the REL board.

The way it was wired in your drawing can cause all kinds of issues as it is completely unclear what voltage levels will be at those GPIO pins.
Either fully isolate it (thus no wires at all between the boards) or at least have the GNDs connected if you're not isolating it.

DrBakterius, 2023-05-16:
Thanks again for the detailed explanation.

Of course I had looked at the table - even for a long time when I had assembled it.

If only one common power supply is used for ESP and relay modules, then booting is no problem. When I used separate power supplies, the ESP only booted when the relay modules were not supplied with power. But I don't understand that, because in principle nothing has changed on the GPIOs. 3V3 goes to VCC on the relay module and the GPIOs go to GND when triggered, which makes the optocoupler (LED) work.

And with complete isolation, only the controls for the optocouplers go from the ESP to the relay modules. At least this must be the case in order to function. However, even in this constellation, the ESP sometimes crashed when switching a GPIO. This is the same as when I use only one power supply. That's why I went back to the original variant.

Is it a problem for the ESP in the longer term if it crashes and restarts sometimes? It is back online within a few seconds.

DrBakterius, 2023-05-16:
> The relais may cause some interference which sets off something in the ESP board.
After all it is a magnetic field and also the contacts may sometimes spark when they do switch some device.

That could also well be the reason. When I have a bit more time again, I'll give it a try. But it may take a while...

DrBakterius, 2023-05-20:
> The relais may cause some interference which sets off something in the ESP board.
After all it is a magnetic field and also the contacts may sometimes spark when they do switch some device.

That was the right hint!

I tested the circuit without a load on the relay. And the resets did not occur. I replaced the connected device (a small pump for my fish tank) with another one (a heater). Since the pump is an indicative load, this may have caused the faults when it was switched off. Now it is connected to a relay that is further away. Initial tests look good. The ESP was not restarted even after at least 20 switching processes.

Many thanks for your support!

TD-er, 2023-05-20:
Good to know.

closed 2023-05-20 by TD-er