letscontrolit/ESPEasy #4669: CCS811 not in love of OLED SSD1306/SH1106 Framed

betux69, 2023-05-15:
Hello,

I used Build: [ESP_Easy_mega_20230508_climate_ESP8266_4M1M May 8 2023](https://github.com/letscontrolit/ESPEasy/releases)
My devices are :
SCD30 i2C: 0x61
SDS011/018/198 : HW Serial0
CCS811 TVOC/eCO2 : I2C 0x5A
 OLED SSD1306/SH1106 : I2C 0x3C
Switch input - Switch

When i put OLED SSD1306/SH1106, CCS811 is disable ad i can't enable it.
When OLED SSD1306/SH1106 is not on the breadboard it works.

when i make a scan I2C with all component i have this 
0x3C	
0x40	
0x5A	
0x61

When i put off OLED SSD1306/SH1106
0x40	
0x5A	
0x61

For me i have 3 devices i2C and not 4.
I don't know who is 0x40.

An other point, i use always on my developpement a buzzer with this rule 
On System#Boot do 
rtttl,14:d=10,o=6,b=180,c,e,g,c,e,g
endon

It doesn't work.
TTL is active with this build ?

Thanks

tonhuisman, 2023-05-15:
We have had to add a limitation on the features of the Climate build, and that results in rtttl being disabled, I'll see if that specific feature can be re-enabled for the Climate build.

For I2C address 0x40 we support these devices: `SI7021,HTU21D,INA219,PCA9685,HDC10xx,M5Stack Rotary enc` 
What exact ESP unit/board are you using? As it may have some on-board I2C device that's not (yet) supported by ESPEasy. Maybe that also interferes with the OLed display, but that's not very likely.

TD-er, 2023-05-15:
Please also try to uncheck on the Tools->Advanced page this setting: "Check I2C devices when enabled"
This will perhaps help with the CCS811 being disabled over and over again.

tonhuisman, 2023-08-05:
This issue can be closed, as it appears to be solved. 😄

closed 2023-08-05 by TD-er