letscontrolit/ESPEasy #4670: Add support TMP107 Temperature Sensor with Daisy-Chain UART, EEPROM, and Alert Function

3dalex07, 2023-05-16:
Please add support for [TMP107](https://www.ti.com/product/TMP107) Temperature Sensor with Daisy-Chain UART, EEPROM, and Alert Function
I find library for arduino may be it help impelement support of  this sensor
 https://github.com/jgromes/SmaartWire

Thanks

TD-er, 2023-05-16:
What makes these sensors specifically more useful compared to the 1-Wire sensors?
What's against them (IMHO) is that they need another serial port, which is a rather limited resource.

It seems though that these devices can be daisy chained, similar to the 1-Wire sensors, but that means a plugin for these needs a shared managing (software) instance to coordinate shared resources.
So they seem to have come up with yet another standard to probably only allow for their own devices to use this bus.
I don't really see why they simply can't use any of the already existing busses out there.

3dalex07, 2023-05-16:
The robust UART-compatible SMAART wire interface transfers data over a single wire at distances of
up to 1000 feet (300 meters) between consecutive devices in the chain, and is capable of communicating in a
daisy-chained configuration with up to 32 devices on a single bus

TD-er, 2023-05-16:
Still it feels a bit like re-inventing the wheel as there are already several busses and bus protocols out there.
For example ModbusRTU, when sent over RS485, is doing something really similar and when loweing the bitrate far enough, can even do upto 1200m.

But I will look into it.

One question popping up in my head about this is whether it needs different configurations/code when used via different interfaces like RS485 or CAN bus as those act differently when addressing devices as far as I thought.

3dalex07, 2023-05-17:
How can i help to implement support of this sensor?

tonhuisman, 2023-05-17:
Sure, here's some documentation to get started https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html
Be warned, it can be a bit daunting at first, but just ask for help where & when needed 👍