letscontrolit/ESPEasy #4671: Web based flasher wrong firmware upload

sp2ong, 2023-05-17:
Wen I use web based uploader for Wemos D1 Mini via Web based flasher and slecet 

20230508 Normal 4M [ESP8266,ESP32,ESP32-S2]

and Connect next screen show me that will be uploaded "CLIMATE 4M" firmware instead "NORMAL 4M"

below screen dumps
[image]
[image]

sp2ong, 2023-05-17:
But when I use link https://td-er.nl/ESPEasy/latest/  
it works OK I select NORMAL 4M and install NORMAL 4M

TD-er, 2023-05-17:
I will have a look.
Probably a mix-up in my script to generate the JSON and HTML

TD-er, 2023-05-17:
Hmm I re-generated the files, can you check again?
I really doubt it is fixed as the diff didn't show anything.
I copied the manifest files and the index.html to a directory "bug" and ran a diff on all:

```bash
root@s01:/var/www/td-er.nl/web/ESPEasy/bin# ls *.json|xargs -n1 -I{} echo "diff {} bug/{}"|sh
root@s01:/var/www/td-er.nl/web/ESPEasy/bin# diff ../index.html bug/index.html
122,124c122,124
<     <br>
<     <br>
<     See <a href="latest/" >latest/</a> for a pre-release test build.
---
> <br>
> <br>
> See <a href="latest/">'latest/'</a> for a pre-release test build.
134c134
< </html>
\ No newline at end of file
---
> </html>
```

sp2ong, 2023-05-17:
Yes, Now is OK, thanks:

[image]

closed 2023-05-17 by sp2ong

TD-er, 2023-05-17:
Hmm really strange to see it now seems to work.
Thanks for testing anyway.