letscontrolit/ESPEasy #4673: Add Hi-Link HLK-LD2410 mmWave human presence sensor

iz8mbw, 2023-05-20:
Hello.
It will be nice to add the HLK-LD2410 24GHz mmWave human presence sensor.
The sensor is cheap and available on AliExpress.

Website and download center:
https://www.hlktech.net/index.php?id=1095
https://www.hlktech.net/index.php?id=download-center

Datasheet and user manual: https://drive.google.com/drive/folders/1p4dhbEJA3YubyIjIIC7wwVsSo8x29Fq-?spm=a2g0o.detail.1000023.17.93465697yFwVxH

An Arduino implementation: https://github.com/ncmreynolds/ld2410
ESPHome implementation: https://github.com/rain931215/ESPHome-LD2410

[image]

Thanks!

TD-er, 2023-05-20:
I just received 4 types  of these 24 GHz modules 2 days ago.
So they are for sure planned to be implemented :)

iz8mbw, 2023-05-20:
Great, we are in sync. :)

chromoxdor, 2023-05-30:
Any advice which module i should buy. (There seem to be at least three versions shapewise)

Wookbert, 2023-09-13:
@tonhuisman
Argh! Sorry for the duplicate (and sorry to myself wasting my own time). Did a search for 2410 and LD-2410 here, but the search engine didn’t come up with any results. Should have used a wildcard search I guess (not even sure whether that’s supported).

Dickesplumpel, 2023-09-15:
There is a support for HLK-LD2410 sensor since Tasmota 12.3.1 . You have to build your own firmware with optional feature "USE_LD2410". Best detection sensor I've ever played around with. But for "professional" use I'm waiting for implementation in espeasy, rules are awesome and offer much more freedom for creativity.

Wookbert, 2023-09-15:
Wish HiLink would be a bit more talented in designing intuitive and appealing Smartphone apps. That HLKRadarTool is kind of awkward to use and understand in every detail.

Dickesplumpel, 2023-09-15:
I have only 2 sensors without bluetooth and so far not tested the tool for serial communication. I only build custom tasmota firmware two days ago with visual studio and this worked really good. I think tasmota "programmed" the LD-2410 an so I could use it with espeasy, the output pin (not rx and tx) from sensor as switch input. I only need motion detection no distance measuring.
But I have no glue whats really going on ;-)

Wookbert, 2023-09-15:
No glue or no clue? 🤪 But seriously: I’m currently also using just the high level signal pin, but a UART connection (Rx/Tx) somewhat makes sense. It enables for instance the use of the luminosity sensor reading which those LD24xx boards provide (yep, what looks like an SMD LED on those boards is actually a luminosity sensor, if I’m not totally wrong). Think of switching on the lights on motion & presence detection, but only at a certain level of darkness.

Dickesplumpel, 2023-09-15:
I often wondered what it has to do with in german "Kleber", so clue makes sense :-). In tasmota 5 values are transmittet over the RX and TX lines. As far as I understand one value is distance of "fast" moving object, one is distance of "slow" moving or nearly static objects, two are power consumtion and the other I don't understand. Andreas Spieß tells in his video that the adjustment is also done over these lines with a special software.

Dickesplumpel, 2023-09-16:
Hi, its on the list, great, I don't want to push! :-)

iz8mbw, 2023-09-16:
Thanks @tonhuisman !

tonhuisman, 2023-09-16:
> Hi, its on the list, great, I don't want to push! :-)

I've ordered a few of those units (just to get into the 12-day delivery category 😛), so they should arrive soon, and I can start coding 💌

Dickesplumpel, 2023-09-16:
Great. When they are “implemented” I will start a project with four to six of these sensors controlling 12 150cm LED-tubes.
Thanks again for this wonderful playground!

TD-er, 2023-09-16:
Given they need a serial port, this may be a bit challenging to mount several to the same ESPEasy node.
I guess using the ESP32-S3 may be the better option here as this one has 3x HW serial and you can still access the console via USB as that's a separate serial port (emulation).

Or you need to add some I2C to UART chip to add more serial ports if you really need extra ports.

Dickesplumpel, 2023-09-16:
I haven’t played around with the software-tool to program these sensors, maybe next week. Maybe they have a ROM to store the settings and then I use only the output pin as switch input. Or two ESP32-S3 with 2 or 3 sensors, each one 1 relay, both relays parallel switching lights. Or every sensor one D1-mini (my favorite 8266), one ESP in AP mode the others connected to this AP, never tried, sounds challenging. Or all ESPs connected to node-red over mqtt, but I like standalone versions only sending messages when something is wrong. Its a project for wintertime and I don't know what possibilities are given by your integration into ESPEasy. I let myself be surpriesed whats possible when I start this project. ;-)

Wookbert, 2023-10-03:
@TD-er Any idea as to when approx. ESPEasy will support the LD2410 via UART (incl. support for its ambient light sensor)?

iz8mbw, 2023-10-03:
I suppose we are waiting for @tonhuisman Ton.

iz8mbw, 2023-10-14:
Ciao @tonhuisman 
The module is delivered here.
[image]

I'm ready to test :-)

tonhuisman, 2023-10-14:
> (incl. support for its ambient light sensor)?

@Wookbert 
Haven't been able to find info on that sensor, other than that this sensor can be used to automatically turn on and off lights, based on human presence. Are you sure that's included, or are you aiming at a specific board?

Wookbert, 2023-10-14:
@tonhuisman 
[image]
That one SMD component on the room facing side has a clear housing, which tell me it is an optical component. Question is: LED or Ambient Light Sensor. Inspecting it under the magnifying glass doesn’t tell me enough. I've however applied 3V3 directly for like half a second to that component (with the board not connected/powered otherwise) and it didn’t light up (tried both polarities).

IMHO it is an ambient light sensor, something like the [PO188](https://aliexpress.com/w/wholesale-PO188.html?spm=a2g0o.home.search.0); LCSC lists hundreds of such similar-looking SMD components if one searches for _Ambient Light_.

The iOS/Android HLKRadarTool App also has a setting for photosensitivity when in Firmware 2.04 (and Engineering Mode?):
[image]

Finally ESPHome seems to have integrated the (ambient) light reading already. Search for the word `light` on [this page](https://esphome.io/components/sensor/ld2410.html) (you’ll find `light function` and `light_threshold`).

Wookbert, 2023-10-14:
@tonhuisman Perhaps you can learn from ESPHome’s Source Code (looks open source to me) how they read the ambient light sensor. Or in worst case simply ask them.

Perhaps you could also be inspired by ESPHome’s GUI integration of the LD2410. I haven’t understood the detection matrix yet, but to my understanding the LD2410 is capable of detecting where in the room someone is present. A very simple graphical representation in ESPEasy UI, rather than just a list of output numbers would be nice. (Sensor in the middle, returned values arranged to that according to their location in the room).

tonhuisman, 2023-10-14:
It must be something undocumented, as I do see that component on my board (I own a similar one), but it's not in the documentation, or the output data, so there is no documented way of reading the data, AFAICS 🤔

Wookbert, 2023-10-14:
Again: Looks like ESPHome (among others?) have managed it. Just check their code. (Sorry about my ignorance, little to no clue about programming).

tonhuisman, 2023-10-14:
Peeked in the ESPHomve code, and it seems it's an undocumented field, but still included in the Engineering mode. Think I'll be able to handle it. 😄

Wookbert, 2023-10-14:
If I recall correctly that 2.04 firmware is even still labelled as experimental. Well, who cares!?

tonhuisman, 2023-10-16:
A first build is brewing on this plugin from this [GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/6538996739).
Who's interested can use that for testing. The plugin is available in the `Collection G` and `MAX` builds, and only tested on ESP32 so far.

Wookbert, 2023-10-16:
But should it theoretically run on an ESP8266?

TD-er, 2023-10-16:
Just looked at the code and I see no reason why it shouldn't run on ESP8266
Just make sure to use hardware serial port as SW serial at 256000 baud isn't a good idea.

tonhuisman, 2023-10-16:
> But should it theoretically run on an ESP8266?

It should, but I just haven't tested that yet, but I will, soon. And I'll also test using an I2C Serial adapter, as supported by ESPEasySerial.

Wookbert, 2023-10-16:
`Just make sure to use hardware serial port as SW serial at 256000 baud isn't a good idea.`
To translate this and to remind myself: Hardware serial means ESP8266’s GPIO 1 (aka TX0) and GPIO 2 (aka RX0), labeled TX and RX on a Wemos D1 mini board layout.

TD-er, 2023-10-16:
Nope, GPIO 1 & 3
See: https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp8266

You can also use GPIO 13 & 15 as "Serial0 swapped", which is essentially Serial0, but then using different pins.
However you may need a PNP transistor for GPIO-15 as this is a special boot strapping pin.

Wookbert, 2023-10-16:
Sorry, 2 was a typo. Fixed. Please delete your „Nope“-comment, I'll remove this one afterwards. Just to keep things clean and avoid confusion to readers.

TD-er, 2023-10-16:
But if you're looking for parts building new projects, I strongly suggest to look for ESP32-S3 or ESP32-C3.
The S3 does have 3 HW serial ports and an USB port so you can still run the ESPEasy console without loosing a HW serial port.
The C3 is available in really small form factors like the Seeed Studio versions with an external antenna and is really snappy due to its RISC architecture.

Wookbert, 2023-10-16:
Still have plenty of Wemos D1 mini clones laying around, which I guess have sufficient performance for presence (& ambient light) only sensors. If you tell me however that the C3 is significantly more power efficient (oh, that electricity bill...) I might consider switching 😃.

TD-er, 2023-10-16:
Just looking at these 24 GHz presence sensor parts and I don't think the ESP is the biggest consumer here.

24 GHz oscillation circuit and low power will not be a great combination for quite some years I guess.

But the ESP32-C3 is quite efficient
I just connected one of those Seeed Studio ones to a USB power meter and it takes between 35 - 40 mA on average. (WiFi is active, but isn't doing much right now)

Wookbert, 2023-10-16:
I know, the LD-2410 (and FMCW sensors in general as I understand) are really power-hungry. 80 mA @ 5V, that’s 0.4 Watts ... 24/7/365. Quite a bit. But at least they work reliably, compared e.g. to the el cheapo RCWL-0516, which creates more problems, than it solves.

iz8mbw, 2023-10-20:
Hi @tonhuisman 
I'm testing the LD2410 and for now I'm focusing on the "Presence" value.
What I have discovered is that ESPEasy took about from 30 seconds to 45 seconds to detect / notice the presence and it took about the same time to detect / notice that the presence disappear.
I don't know if this delay is a normal behavior of the sensor of the library, but IMHO it's a big delay.

More please tell me/us what is the best "Interval" value to set, for know I set "Interval" to 1 second but on the log I don't see any "polling" made every 1 second.

Thanks!

tonhuisman, 2023-10-20:
> but on the log I don't see any "polling" made every 1 second.

An event, and log, is only generated if any of the values is different from the previous value, so if nothing changes, nothing is sent/logged. (I'll add a READ log on every interval, that's probably helpful/reassuring during testing.)

> took about from 30 seconds to 45 seconds to detect / notice the presence

It may depend on the physical installation how accurate/responsive the sensor is, but your findings seem to be _very_ slow. I've picked the sensor up (not mounted in any case, just the sensor on the wires), turned the antenna towards myself and the presence detection was nearly instant, most of the time, but sometimes it lagged a bit (a few seconds).

I've seen that the Presence pin on the sensor also has the current state, and I've connected that to a GPIO and Switch task, and it is turned off after 5 seconds (sensor default value) of stationary presence has passed (and no moving presence has happened). This is a setting that I intend to expose for configuration.

What kind of ESP did you connect it to, ESP8266 or ESP32?

iz8mbw, 2023-10-20:
Also using the sensor not covered but it's seems slow. 
I'm only using TX and RX pins so not suing also OUT pin.
The board I'm using is ESP32 (16 MB).

I have another LD2410 and I'll test also this.

Wookbert, 2023-10-20:
> but it's seems slow.

Can you describe in detail when, where, in which situation exactly it is slow? Is the response on detecting the very first motion slow or is it slow on updating, respectively detecting the „end of motion“?

Because if it slow at the end of motion, it could be the OFF signal delay, which tonhuisman mentioned to be 5 sec by default (which in his current firmware version not yet user adjustable [though it should be], but perhaps there is a bug).

tonhuisman, 2023-10-20:
The default delay is 5 seconds, but if the setting is changed using other software, f.e. via Bluetooth, then it _may be_ that the settings are saved in the sensor (not really sure about that, also working on another sensor that _does_ keep the last settings... 🤔)

iz8mbw, 2023-10-21:
Hello @tonhuisman, hello @Wookbert, hello all.
I have connected also the "OUT" pin so now there is a more clear scenario.

LD2410 connected to ESP32 board on "HW Serial 1" and "OUT" pin connected to a "Switch input - Switch" as "Normal Switch".

This is what happens after a board reboot and when I'm in front of the sensor:
[image]

as you can see, "OUT" pin works about in real time but all the other data via Serial are still all "0".
So, here starts the delay, "OUT" pin switch to high and data in Serial are still not "transmitted" to the ESP32 board.

Then, after some seconds when I'm still in front of the sensor, data via Serial arrives but as you can see "Presence" is still "0" but other data are populated:
[image]

Then, after some other seconds when I'm still in front of the sensor, "Presence" starts to be "1" and other data are refreshed:
[image]

Then, after some other seconds, I go away so sensor does not recognize me and as you can see "OUT" pin switch correctly to "0" but the rest of data in Serial are still populated and also "Presence" is still to "1":
[image]

Then, after some other seconds, I'm still away so sensor still does not recognize me and as you can see "OUT" pin still remain correctly to "0" but the rest of data in Serial are still populated and refreshed (compared to before) and also "Presence" still remains to "1":
[image]

Then, after some other seconds, I'm still away so sensor still does not recognize me and as you can see "OUT" pin still remain correctly to "0" and now finally "Presence" switch to "0" but other values are still populated (you cannot have a Presence to "0" and a Distance to "451"):
[image]

So, based on my tests and results, the Serial output is very delayed.
Also strange to have a "Distance" to 451 if I'm away from Sensor, and if I still remain away from Sensor, "Distance" did not refresh and come back to "0".

tonhuisman, 2023-10-21:
@iz8mbw Do you have the sensor powered at 5V? As that's a requirement of the chip, AFAICS. The serial signals should still provide the regular 3.3V signals, according to the documentation, so no level converter should be needed.

iz8mbw, 2023-10-21:
Yes 5V.
Tried also 3.3V but it's the same.

tonhuisman, 2023-10-21:
I'm partially rewriting the ld2410 library, making it work (more) asynchronous. It seems it's not returning data in time when polled. Not sure when that's completed though 😅

iz8mbw, 2023-10-21:
Great!

Wookbert, 2023-10-21:
@iz8mbw Can you ensure that your described delays aren’t caused by some hick-ups in the serial communication (e.g. wrong serial settings, if there are any)?

Dickesplumpel, 2023-10-21:
I watch the same/similar behavior like iz8mbw over mqtt to node-red, switch on output pin reacts imediatly and sends mqtt message immediatly, distance hangs in my case for hours at 431 even nobody is in the room.

iz8mbw, 2023-10-21:
> @iz8mbw Can you ensure that your described delays aren’t caused by some hick-ups in the serial communication (e.g. wrong serial settings, if there are any)?

Nhaaa

Dickesplumpel, 2023-10-21:
I have a LD2410C, maybe thats causing trouble.

iz8mbw, 2023-10-21:
I'm also using/testing LD2410C model.

tonhuisman, 2023-10-21:
Besides trying to move to a more async ld2410 library, I've made quite nice improvements already by reading the sensor data in the 50/sec function, instead of the 10/sec function (where the rest of the initialization is done for timing reasons).
I'll update later today, as I'm also working on modifying the sensor sensitivity settings 😉 

Please be aware that when using the sensor without any 'directional shielding', it ~seems to~will detect also human presence _behind_ the intended antenna viewpoint, i.o.w,: it is not really directional unless you shield the direction you don't want to detect... 🤔

iz8mbw, 2023-10-21:
@tonhuisman Ton what about this other library?
https://github.com/shabaz123/LD2410

tonhuisman, 2023-10-21:
> @tonhuisman Ton what about this other library? https://github.com/shabaz123/LD2410

That's not a library, that's a simple demo/PoC to show the sensor works, in Arduino and Python code. Not usable, and incomplete.

Dickesplumpel, 2023-10-21:
And from tasmota?

tonhuisman, 2023-10-21:
> And from tasmota?

??

AFAICS, Tasmota has implemented the protocol directly in their driver, and should work similarly to the ld2410 library. That driver is not usable from ESPEasy, as both use a completely different plugin-architecture.

Dickesplumpel, 2023-10-21:
I have no clue about that, it was just an idea that parts of it could be copied.

tonhuisman, 2023-10-21:
The ncmreynolds/ld2410 library I'm using (as linked by iz8mbw, first message) is working pretty good (after applying an open PR and some more suggestions), but it does have the 'issue' that it is blocking for some (short) time when sending a command to the sensor (waiting for command acknowledgement). This somewhat interferes during initialization with the regular handling by ESPEasy, that I want to avoid. (Several possible solution-paths that I'm still thinking about)

For reading the regular data we don't need to send any command, it just bursts out the data continuously. For this data-reading to work successfully, it has to be read at a high speed, and as the default baudrate is 256000, it's easy to loose track, and that's what is currently the case.
I think I've fixed that locally, but want to combine those fixes with other improvements before updating the PR.

tonhuisman, 2023-10-21:
Updated a lot of stuff, should perform better now. Downloads available from [this GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/6599540178) once that's finished.

iz8mbw, 2023-10-21:
Built failed

[63 more replies]

PR #4852 (2023-11-27): [P159] Add LD2410 Radar presence detection
Resolves #4673 

P159 - Presence - LD2410 (24 GHz Radar human presence detector)

Using a _beta_ version of [nvmreynolds/ld2410](https://github.com/ncmreynolds/ld2410) library, with an open PR included to have Engineering mode, and other experimental patches like the Ambient Light sensor support.

Features:
- All presence related measurements available from the sensor
- All Engineering measurements available from the sensor
- Ambient Light value (undocumented feature in Engineering mode, only available on boards that have the AL sensor)
- Output pin s [... cut]
closed 2023-11-27 by TD-er