letscontrolit/ESPEasy #4675: SD Card detected but not possible to use system or value logging

305651, 2023-05-23:
I compiled a custom image with SD Card support and tried to use logger feature. I configured SPI and enabled both system log to SD and value loggger but no files are created on SD. When I copy some files to SD, they are visible via "Browse files on SD" feature.

### Steps already tried:
Tested all connections between ESP32 and SD with ohm-meter
Re-flashed firmware with full chip erase
Tried 3 different FAT32-formatted sd-cards

### Expected behavior
VALUES.CSV is created on SD and measurements are logged. Also log.dat is created when system log to SD is enabled.

### Actual behavior
No files are created on SD

### Steps to reproduce
Compile firmware with SD Card support
Connect SD to ESP32
Configure SPI and enable data logger

### System configuration
I am attaching config.dat, custom.h and compiled firmware image. [files.zip](https://github.com/letscontrolit/ESPEasy/files/11547439/files.zip)
Screenshots with SD Card related settings:
[image]
[image]
[image]

Hardware:
ESP32-DEVKITC-32D V4 (https://kamami.pl/esp32/573333-esp32-devkitc-32d-v4-plytka-rozwojowa-iot-z-modulem-wi-fi-esp-wroom-32d.html)

I also looked in documentation, but unfortunately there is little information about SD Card support. I found this topic on forum: https://www.letscontrolit.com/forum/viewtopic.php?p=41471&hilit=csv#p41471 and according to it, logger function should work properly. In devices tab, measurements are displayed properly and they are being refreshed every second as I configured. Please inform me if You need more information about this issue. I am sorry if I made any dumb mistakes, I am not a software developer.

305651, 2023-05-23:
It was quite old SD, 1 GB, I tried also with 2 GB with same result. I can check tomorrow the exact model as I left it in my university, I am attaching a photo, but SD is barely visible.
[image]

TD-er, 2023-05-23:
Maybe you can also try to format the SD card using FAT and not vFAT/FAT32?
Just FAT can handle upto 2 GB.

I don't know if there's a difference per SD card whether they support all different data modes (e.g. 1/2/4 data lines)

Another thing is that you're now using the pre-configured set of HSPI.
Maybe also try to use VSPI (or a custom setup)?
The one you're now trying is using GPIO-12 which is a big red flag.
The pin state of GPIO-12 during boot determines the flash voltage.
So either it will make the ESP run highly unstable or fry your flash chip if the pin state is wrong at boot.
This also means there is a pull-up or -down resistor present to set the proper boot state of that pin.
Perhaps this does affect the SPI bus too?

TD-er, 2023-05-24:
I've been thinking a bit more about this...
Things to test/try:
- Use SD cards as old as possible
- Format using FAT, not FAT32/vFAT
- Check to see if the supplied voltages are what they should be
- Add capacitors over the power line close to the SD-card reader (e.g. 10 uF and 100 nF, with the latter one as close as possible to the module)
- Check on a PC to see if the file system may be corrupted. (and properly unmount the SD card before ejecting)

TD-er, 2023-05-24:
What is powered from the 5V line?
Some devices use voltage regulators that need quite a bit of voltage drop to function stable.

Since it is reading when the SD-card is formatted with FAT32, I guess we can rule out SD-card and filesystem compatibility.
Remains the question about voltages and.... (extremely unlikely ;) ) some bug in ESPEasy...

To be honest, I have not tested/used the SD card features myself, so I don't know if or when thing might be broken in ESPEasy code.
All I know is that there hasn't been any reported issue with SD card use recently. So either it should just work, or nobody is using it.

tonhuisman, 2023-05-24:
Have you tried to upload a file to the SD card from the ESPEasy Tools page? If that works, then we know that at least the hardware is working as intended...

TD-er, 2023-05-24:
> Have you tried to upload a file to the SD card from the ESPEasy Tools page? If that works, then we know that at least the hardware is working as intended...

Doh!
Why didn't I think of this suggestion ....  Need more coffee...

305651, 2023-05-24:
I cannot find such feature, there is no upload button on SD card page, only on internal fs file browser

TD-er, 2023-05-24:
Hmm I guess one of us (probably either @tonhuisman  or me) has to really test the SD card functionality ourselves.
Or maybe someone else is actually using this SD card feature.

Just to be sure it isn't something broken in the (somewhat) recent change to the Espressif IDF 4.4 SDK, could you test a build of roughly 2 year ago? (not sure when I updated it, but I don't think it has been 2 years)

305651, 2023-05-24:
Is there any pre-compliled known working build with SD support? I only found one on the forum, in the link I posted earlier but dropbox link is expired. If no, I will try to compile myself but I will be able to test it next week, probably on monday.

TD-er, 2023-05-24:
OK, some more info, found [here](https://randomnerdtutorials.com/esp32-microsd-card-arduino/#comment-571355) 

Can you try this FAT32 formatter tool?
http://ridgecrop.co.uk/index.htm?fat32format.htm

TD-er, 2023-05-24:
OK, @tonhuisman found some issues and is working on a fix :)

305651, 2023-05-24:
> OK, some more info, found [here](https://randomnerdtutorials.com/esp32-microsd-card-arduino/#comment-571355)
> 
> Can you try this FAT32 formatter tool? http://ridgecrop.co.uk/index.htm?fat32format.htm

I will try on monday, currently I don't have access to my equipment. For formatting I used "Rufus" tool and Windows built-in formatter, cards formatted with any of those tools worked properly with Arduino's SD Card example.

tonhuisman, 2023-05-24:
Available once [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/5073658783) is finished, if you could download that and test, please?

305651, 2023-05-24:
Currently I don't have access to my ESP32, but I can ask my supervisor to flash this image to ESP and check if files are created. Thank you for quick support and solving this problem. One more quesion, I found a "proof of concept" SD logger plugin, is it possible to integrate it into current firmware? Here is the link: https://www.letscontrolit.com/forum/viewtopic.php?t=2087

tonhuisman, 2023-05-25:
I've added an Upload feature to the SD Card filebrowser, and fixed some display misalignment. Downloads available from this [Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/5084792930).

> found a "proof of concept" SD logger plugin

We're looking for another solution for that, _possibly_ by extending the [Cache Controller](https://espeasy.readthedocs.io/en/latest/Controller/C016.html#c016-page) with support for storing files on the SD Card. That will be a separate PR, if and when it's resolved.

305651, 2023-05-26:
Thanks for this, I will try it on monday and report back if it works correctly.

tonhuisman, 2023-05-26:
Squashed another bug in the SD Card filebrowser, available when [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/5094601004) is finished.

305651, 2023-05-29:
Tested ESP_Easy_mega_20230526_custom_ESP32_4M316k_LittleFS.factory.bin, log.txt is created, upload to sd works, I cannot test value logger as this build does not have support for ADS1115. I will try to compile my own build and test it.

305651, 2023-05-29:
I will try soon with different plugin and report back. 
I have a problem compiling, I get this error:
[image]

305651, 2023-05-29:
Logger is working, values.csv is created and it contains data, I used "dummy device" plugin

305651, 2023-05-29:
> I will try soon with different plugin and report back. I have a problem compiling, I get this error: [image]

Nevermind, it works now

TD-er, 2023-05-29:
That build error will often happen when you build it first time (or right after a full clean) or when you unfold another env. in PlatformIO while it is building.

305651, 2023-05-29:
Yes, it was exactly when I started build for a first time. Now I compiled build with ADS1115 support and logger works properly. Thank you for support and fixing this issue.

PR #4676 (2023-05-29): [SD] Corrections for writing logging to sd-card
Resolves #4675 

Bugfixes for:
- Logging wasn't written to SD-card on ESP32.
  - Change: Now using `log.txt` instead of `log.dat`.
- VALUES.CSV wasn't written to SD-card on ESP32.
- File sizes where not readable in SD filebrowser.
- Only protected (ESPEasy system-)files could be deleted from SD-card (inverted check).
- Fix SD Card filebrowser display misalignment
- Add Upload feature for SD Card filebrowser
- Fix traversing subdirectories in the SD Card filebrowser

Testing:
- [x] ESP32 (on-board card connector)
- [x] ESP8266 (external card-reade [... cut]
closed 2023-05-29 by TD-er