letscontrolit/ESPEasy #4677: Wing - rotary encoder feature request - monitor volume

AHunty, 2023-05-28:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

tonhuisman, 2023-05-28:
Some more detailed information is required.

TD-er, 2023-05-28:
Yep, what Ton said, I really can't see what the actual feature request is.
We do have support for a rotary encoder.
See: [P059 documentation](https://espeasy.readthedocs.io/en/latest/Plugin/P059.html#p059-page)

tonhuisman, 2023-05-28:
And also a couple of I2C connected ones, see [P143 documentation](https://espeasy.readthedocs.io/en/latest/Plugin/P143.html#p143-page)

tonhuisman, 2023-08-05:
This issue should probably be closed, as there is no reply or updated info.

TD-er, 2023-08-05:
Yep, no need to keep issues open if there's no reply from the original poster.

closed 2023-08-05 by TD-er