letscontrolit/ESPEasy #4678: Inverted "Force WiFi B/G" behaviour

chemmex, 2023-05-28:
(almost) recent official normal ESP8266 build. With "Force WiFi B/G" is **unchecked**, the device connects in 802.11g mode:

```
INIT : Booting version: ESP_Easy_mega_20230507_normal_ESP8266_4M1M, (GitHub Actions) mega-20230507 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
704 : Info   : INIT : Free RAM:28168

797 : Info   : Set Network mode: WiFi
901 : Info   : WIFI : Set WiFi to STA
1005 : Info   : WiFi : Start network scan channel 2
1069 : Info   : WiFi : Scan finished, found: 3
1071 : Info   : WiFi : Best AP candidate: AAA 33:FF:22:79:1F:11 Ch:2 (-40dBm) WPA2/PSK
1072 : Info   : WiFi : Added known candidate, try to connect
1073 : Info   : WiFi : Best AP candidate: AAA 33:FF:22:79:1F:11 Ch:2 (-40dBm) WPA2/PSK
1086 : Info   : INIT : Free RAM:24920
1174 : Info   : INFO : Plugins: 47 [Normal] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
1175 : Info   : EVENT: System#Wake
1286 : Info   : WiFi : Best AP candidate: AAA 33:FF:22:79:1F:11 Ch:2 (-40dBm) WPA2/PSK
1287 : Info   : WIFI : Connecting AAA 33:FF:22:79:1F:11 Ch:2 (-40dBm) WPA2/PSK attempt #0
1289 : Info   : WIFI : Set to 802.11g
1387 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
1391 : Info   : Webserver: start
1404 : Info   : EVENT: System#Boot
1427 : Info   : EVENT: WiFi#APmodeDisabled
```

When unchecked:
```
INIT : Booting version: ESP_Easy_mega_20230507_normal_ESP8266_4M1M, (GitHub Actions) mega-20230507 (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
707 : Info   : INIT : Free RAM:28168

800 : Info   : Set Network mode: WiFi
904 : Info   : WIFI : Set WiFi to STA
1008 : Info   : WiFi : Start network scan channel 2
1072 : Info   : WiFi : Scan finished, found: 3
1075 : Info   : WiFi : Best AP candidate: AAA 33:FF:22:79:1F:11 Ch:2 (-39dBm) WPA2/PSK
1075 : Info   : WiFi : Added known candidate, try to connect
1076 : Info   : WiFi : Best AP candidate: AAA 33:FF:22:79:1F:11 Ch:2 (-39dBm) WPA2/PSK
1089 : Info   : INIT : Free RAM:24920
1177 : Info   : INFO : Plugins: 47 [Normal] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
1178 : Info   : EVENT: System#Wake
1289 : Info   : WiFi : Best AP candidate: AAA 33:FF:22:79:1F:11 Ch:2 (-39dBm) WPA2/PSK
1291 : Info   : WIFI : Connecting AAA 33:FF:22:79:1F:11 Ch:2 (-39dBm) WPA2/PSK attempt #0
1292 : Info   : WIFI : Set to 802.11n Force B/G mode
1386 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
1389 : Info   : Webserver: start
1403 : Info   : EVENT: System#Boot
1426 : Info   : EVENT: WiFi#APmodeDisabled
```,
and the connection is made in 802.11n

tonhuisman, 2023-05-28:
The code in that area is quite, well, 'intriguing' (like many of the code in the area of WiFi related stuff), but most likely it's a logic error around the check for that setting 🤔

Last change to that code was 8 months ago, according to git...

TD-er, 2023-05-28:
Hmm this means I have to dive into this 'intriguing code'.... again :(

chemmex, 2023-05-28:
I think it may somehow be connected to #4655 as well. Because now the code behaves like "Do Not Start AP" is always checked.

chemmex, 2023-05-29:
Yes, this is exclusively a ESP82xx problem

TD-er, 2023-05-29:
It might be that I messed up this setting when working on making the 1M Sonoff Basic units work stable.
I think it now tries to first connect using b/g mode anyway and if that fails, it will immediately retry using n mode.

chemmex, 2023-05-29:
To be honest, I can't see n connections since don't remember which build. And no, it doesn't try to connect in n mode anymore, it always stay in g sometimes falling back to b.
Unless if I uncheck Force B/G, of course

PR #4863 (2023-10-30): [WiFi] Fix force B/G mode on ESP8266
Also add automatic fallback to a mode supported by the access point when settings force a mode not supported by the access point.  (only on ESP32 or ESP8266 SDK3.x)

Fixes: #4859 
Fixes: #4678
closed 2023-10-30 by TD-er