letscontrolit/ESPEasy #4684: [Ethernet] Add support for W5500

TD-er, 2023-05-30:
ESP32 classic is the only one with native RMII support for various Ethernet chips like the popular LAN8720A.
Major drawback is the high pin count needed to support RMII Ethernet chips.

The [W5500](https://www.wiznet.io/product-item/w5500/) is available in really small boards already mounted to the RJ45 connector and it only needs 6 GPIO pins to offer network.
This chip already has a running TCP/IP stack.

This should make it possible to add Ethernet to all other ESP32 boards like the ESP32-C3/S2/S3 (and maybe ESP8266 too?) 

- [Overview of various Ethernet chips](https://pcbartists.com/design/embedded/esp32-ethernet-phy-chips-supported-by-arduino-esp-idf/)
- [Espressif IDF Ethernet Example](https://github.com/espressif/esp-idf/tree/903af13e84/examples/ethernet/basic)
- [Espressif ESP32-S3 Ethernet API reference](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/network/esp_eth.html)

Jason2866, 2023-11-12:
Possible with Esp32 Arduino 3.0. SPI PHY are supported.

TD-er, 2023-11-12:
Yep, that's why I have been working on adding ESP-IDF5.1 support for the past 5 weeks :)

Glad to hear you're eager to see it supported in ESPEasy @Jason2866 ;)

Jason2866, 2023-11-12:
@TD-er The other great project ;-)

closed 2024-03-14 by TD-er