letscontrolit/ESPEasy #4686: custom build with DFPlayer no longer allows a variable pin (e.g. D6).

Sminter4, 2023-06-01:
I built an ESP buzzer in 20210223, which unfortunately kept crashing. In the hope of eliminating the bugs of the time, I built my own build today. Exclusively with the required plugins.

Back then, you could simply assign a pin to the DFPlayer plugin. With the current version, that doesn't seem to work anymore. In addition, my ESP is installed in such a way that unfortunately I can no longer use any other pin.

How do I get pin D6 configured via the current build?
A buzzer without sound is useless...

[image]
---
[image]

tonhuisman, 2023-06-01:
As you may have seen in your screenshot, the Software Serial option, that allows to configure (nearly) any RX/TX GPIO pin for serial communication, is missing from the list.
That's actually a (small) bug, you can patch it for now in the file `src/src/CustomBuild/define_plugin_sets.h` around line 2218 (the long line with all the `|| defined(USES_..)`, where a check for plugin P065 should be added: `|| defined(USES_P065)` to avoid Software Serial to be disabled (that's added for reducing the .bin size).
After a new build (assuming you're using the VSCode/PlatformIO setup) and the update uploaded you should have that SWSerial option available again.

tonhuisman, 2023-06-01:
PS: I like that colorful contraption 😄

tonhuisman, 2023-06-02:
> a check for plugin P065 should be added: `|| defined(USES_P065)`

This fix has been merged into `mega` now, so if you update your local sources and rebuild your Custom build it should work as intended.

Sminter4, 2023-06-02:
Thanks for the tip. I have changed the file `src/src/CustomBuild/define_plugin_sets.h` to `|| defined(USES_P065)` added and reloaded everything.
Unfortunately without success. The overview says `SoftwareSerial` but the dropdown menu doesn't allow it.

[image]
---
[image]

Do I maybe need to add something else?

TD-er, 2023-06-02:
With the latest source it should work just fine.
What plugins do you need? And what kind of build? (4M ESP8266?)

TD-er, 2023-06-02:
I just installed "Collection A" build from the [latest build on the web flasher](https://td-er.nl/ESPEasy/latest/)
And this build allows me to select SW serial:
[image]

Sminter4, 2023-06-02:
Yes, exactly this: custom_ESP8266_4M1M

- Analog input - internal
- Generic - Dummy Device
- Switch input - PCF8574
- Display - 7-segment display
- Notify - DFPlayer-Mini MP3 [TESTING]
- Output - NeoPixel (Basic)

Sminter4, 2023-06-02:
Thanks for your build - that made it work! 😄

I probably haven't set all the settings outside of the plugins yet...

closed 2023-06-02 by TD-er