letscontrolit/ESPEasy #4689: Add support for Bosch BMP3xx Pressure sensors

iz8mbw, 2023-06-04:
Hi!
Is it possible to add Bosch Sensortec BMP3xx series Pressure sensors in ESPEasy?
Sensors are BMP380 / BMP384 / BMP388 / BMP390

Docs and Libraries:

https://github.com/boschsensortec/BMP3-Sensor-API
https://github.com/adafruit/Adafruit_BMP3XX

Datasheets:

[BMP384](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bmp384-ds003.pdf)
[BMP388](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bmp388-ds001.pdf)
[BMP390](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bmp390-ds002.pdf)

Many thanks!

TD-er, 2023-06-04:
Hmm, haven't seen those before.
I will order some too.

iz8mbw, 2023-06-04:
https://a.aliexpress.com/_Evle5bn

TD-er, 2023-06-29:
As we speak I am labelling all stuff I received yesterday from the postman.
Included are these sensors.

iz8mbw, 2023-06-29:
Ok, I will receive a BMP390 next week so I can support for testing.
Thanks!

iz8mbw, 2023-07-06:
Hi @TD-er the BMP390 is now delivered here, when you are ready I can support for testing.

iz8mbw, 2023-08-30:
Hi @TD-er any news about implementing these sensors series?

TD-er, 2023-08-30:
Well I can already see the color of my floors and even parts of the original color of my anti-static mats on my desks.
So the re-arrangements of my office is near to "good enough. After which implementing sensors is on my todo list.

TD-er, 2023-09-06:
OK, ADS1x15 is done, so this is next :)

iz8mbw, 2023-09-06:
Great!

TD-er, 2023-09-06:
I've just looked into the datasheets of both BMP388/390 and the already supported BMP280.
The only similarities between those is the brand and the fact they measure temperature and air pressure...
Oh and they seem to use electricity and interface via I2C.

They are so completely different, that I will allocate a new plugin ID for this BMP3xx series.

iz8mbw, 2023-09-06:
Yes, BMP3xx series is different from the already supported BMP280, this is why I asked on this issue to support BMP3xx series.
Thanks.

TD-er, 2023-09-06:
Just started with just using the Adafruit library, to get an idea of how this sensor may work.

But I'm a bit surprised it only took like 30 minutes to write this.  :)
See PR #4789 

not yet tested, but I see no reason why this shouldn't work.
Let's summarize it as "it compiles".

Now I have to dig up some of the ordered modules as I'm sure they have to be somewhere....

TD-er, 2023-09-06:
Hmm that was easy....

[image]

[image]

iz8mbw, 2023-09-06:
Oh.. Yes!
It works well also here, thanks!
[image]

Please can you add the "Altitude" on the "Device Settings" likes it was done for BME68x, for BMx280 and for BMP085/180?

[image]

TD-er, 2023-09-06:
Like this for the BME280?
[image]

Not sure if temperature offset is really needed here, as it is used in the BME280 to compensate for the measured humidity (and offset in temp of course, but that's easy to do in the formula field)

iz8mbw, 2023-09-06:
Indeed, just asking for "Altitude" only.
See also BMP085/180.
'Altitude" is a value (in meters) that should be used to "normalise" pressure to sea level.
So it should be a easy formula to increase pressure value when the sensor is on a specific altitude.

TD-er, 2023-09-06:
So like this?

[image]
Pressure at 0 m altitude is roughly 1021.47656 mbar.

N.B. I also added the detected sensor type.
Since the library only shows 0x50 for BMP3_CHIP_ID and 0x60 for BMP390_CHIP_ID, I can only show "BMP38x" or "BMP390".
If there is another value, it will show just the number, so we could add it to the code.

iz8mbw, 2023-09-06:
Thanks, I will test tomorrow.

iz8mbw, 2023-09-07:
Hello and thanks!
@TD-er just tested latest commit and "Altitude" works as expected, also "Detected Sensor Type" works:
[image]

Just a note, also in this new Plugin returning "nan" when I disconnect VCC or I2C signals (between the sensor and the ESP32) is not covered.
As you know to recognize a the read error returning "nan" is better then returning 0 values to both Temperature and Pressure.
For example for Plugin SHT3x and SHT4x the read error returning "nan" was covered, see comments here https://github.com/letscontrolit/ESPEasy/issues/4590

Apart to manage read error (returning "nan" instead to return 0) for me this PR can be merged.

Thanks.

TD-er, 2023-09-07:
Well I can't yet merge it as we're missing 1 important aspect of the PR...
The documentation.

Good to know the BMP390 is also working as expected, as I never received the BMP390 I ordered.
I only have 3 pieces of BMP388.

I will think about the NaN option...
Not sure if I will add it right now, as we were planning to make it a generic option for all plugins to have an error state value which the user can set.
So better not create a legacy setting where there may be users who have it setup based on plugin specific error states.
Also the default setting is to disable plugins which cannot access the I2C device, so with that option checked you would not get any values if the sensor gets disconnected.

iz8mbw, 2023-09-07:
OK.
About NaN, OK to for the future generic option, make sense, I agree.
And when you will work on the generic NaN option make also sense to remove the disable plugins which cannot access the I2C device.
A temporary (accidental) read error must NOT disable the plugin!

tonhuisman, 2023-09-07:
> A temporary (or accidental) read error must NOT disable the plugin!

The check does actually probe the device for accessibility via the I2C bus, and only after 10 failures in a row, it is disabled.
This is a security/stability measure, to stop failing I2C sensors from blocking the entire ESP, as has been seen several times in the past.

closed 2023-09-07 by TD-er

iz8mbw, 2023-09-07:
Ok but if I have a sensor with 10 seconds of polling and I accidentally disconnect a signal cable and I notice it after some hours I will have the sensor marked as disconnected and then I must manually re-erbale it.
Personally I don't like this behavior.

For me, by default, a sensor must remain enabled until the user will go to disable it and ESPEasy should manage the "read error" (or disconnection) with nan.
In this way, with "nan", the user can understand there is an issue and so solve the issue.

Thanks.

TD-er, 2023-09-07:
You can uncheck this I2C check.
And like I said, we're working on a generic "error state" setting for all plugins.
So that will be added later, not yet for this plugin.

iz8mbw, 2023-09-07:
The sensor will become disabled also if I2C check is disabled.

TD-er, 2023-09-07:
Ah yep, if `PLUGIN_INIT` returns false, it will be disabled.
That's when the sensor is not found.

But that's done with many more plugins.

I have to dig a bit more into the code to see if we added a 'retry' later, as I'm sure that option has been discussed before.

iz8mbw, 2023-09-07:
Ok, for now as told, the sensor will become disabled also if I2C check is disabled and this is annoying.

TD-er, 2023-09-07:
> Ok, for now as told, the sensor will become disabled also if I2C check is disabled and this is annoying.

And that exact part is what I'm looking into right now.

So if you will double check your solder joints, I will double check the code and together we will end up with something that's working as expected :)

iz8mbw, 2023-09-07:
Great and thanks!

PR #4789 (2023-09-07): [BMP3xx] Add support for Bosch BMP384/388/390 temp/pressure modules
Just using the basic Adafruit library as a start.

Fixes: #4689 

ToDo:

- [x] Test with real hardware
- [ ] Strip out Bosch/Adafruit library
- [ ] Add user configurable settings and filters
- [x] Document