letscontrolit/ESPEasy #4690: ESP8266_1M and P005 P021

tomatos-st, 2023-06-04:
I have several Sonoff Basic modules that I use as air/cool controllers,
Recently I wanted to update them (some from 2018)
I can't find a compilation that would have plugin P005 (DHT22) and P021 (Regulator - Level Control) for ESP8285.
In the descriptions there is information that these plugins are now in CLIMATE, but there is no set for 1M
It would be possible to make a simple compilation under 1M containing basic plugins for devices of this type (Sonoff)
e.g. P001,P004, P003,P005,P006,P021?

Thanks

TD-er, 2023-06-04:
These should all be present in the 'normal' builds.

As you can see on the [web flasher page](https://td-er.nl/ESPEasy/latest/), there is an 1M version for the "Normal" set.

You need to flash it via serial anyway as there is no way to perform OTA on an 1M build anymore. It is just too small for OTA updates.

tomatos-st, 2023-06-04:
I searched in https://github.com/letscontrolit/ESPEasy/releases, in the ESPEasy_mega_20230508_ESP82xx_binaries.zip archive there are no 'normal' builds for 1M version.
Your web flasher is great ;).

Once every few years I do a review of the installation, so I can flash it by serial.

Topic is closed
Thanks

closed 2023-06-04 by tomatos-st

tonhuisman, 2023-06-04:
> I searched in https://github.com/letscontrolit/ESPEasy/releases, in the ESPEasy_mega_20230508_ESP82xx_binaries.zip archive there are no 'normal' builds for 1M version.

@tomatos-st Shortly after that release, @TD-er did a great effort to reduce the build-size for these flash-limited 1M builds by using `float` variables instead of `double` for most of the Rules processing. This has the positive effect of more 1M builds succeeding, so these are available again in the web flasher page 😄