letscontrolit/ESPEasy #4697: Notifications - Add Pushover support

iz8mbw, 2023-06-10:
Hello.
Is it possibile to add Pushover into Notifications?

Pushover provides real-time notifications on Android, iPhone, iPad, and Desktop (Android Wear and Apple Watch).

Info and libraries:

https://github.com/brunojoyal/PushoverESP32

https://pushover.net/api

https://support.pushover.net/i44-example-code-and-pushover-libraries

https://randomnerdtutorials.com/esp32-pushover-notifications-arduino/

Thanks!

tonhuisman, 2023-06-10:
There is already an ongoing (though rather silent) thread on this subject in #4472, for the short term the suggested alternative is to use [ntfy](https://ntfy.sh), that supports HTTP and can be used from rules since we introduced the `PostToHTTP` and `PutToHTTP` commands.

Also, we have some technical limitations for any of these notification services requiring HTTPS, most important one being that we don't yet support SSL/TLS connections (still an open pull request, only for MQTT for now), and that it will be quite hard to add support for ESP8266 for SSL/TLS.

iz8mbw, 2023-06-10:
Ok, so since TLS is not implemented (or not yet) Ok to think to use ntfy.
Going to close this Issue with the hope ntfy can/will be implemented.
Thanks

closed 2023-06-10 by iz8mbw

tonhuisman, 2023-06-10:
> with the hope ntfy can/will be implemented.

The suggestion is to use `PostToHTTP` and/or `PutToHTTP` commands from Rules to send out notifications via ntfy, as [demonstrated here](https://www.letscontrolit.com/forum/viewtopic.php?t=9434) and via [easy2ntfy](https://www.letscontrolit.com/forum/viewtopic.php?t=9564) (both are links to the forum)