letscontrolit/ESPEasy #4699: Lolin S2 Mini programming

s0170071, 2023-06-11:
Just wanted to give a heads up for users of the purple S2 mini v1.0.0 board. 

[image]

These were giving me a very hard time programming. 
I finally succeeded using
- windows and the latest esp [tool v3.9.4 ](https://www.espressif.com/en/support/download/other-tools?keys=&field_type_tid%5B%5D=624). Use the S32 bin file with "factory" in the file name. Start address is 0. Press both buttons, release reset first, then the other. Only then the port will show up.
- on Linux (mint) I managed to use the Arduino IDE but only if you quickly and timely reset the board after you hit the upload button. 

What was not working (but did with other boards) was TD-ERs upload webpage. Still nice job, @TD-er :-)
Neither did the ESP Tool on linux, any other prior verions of the ESP Tool on Windows or the Arduino IDE on Windows. 

I like the form factor of these S2 mini boards but I would not recommend buying them. Espressif seems to get lost in their various chip versions lately.

tonhuisman, 2023-06-11:
Have you tried the Latest build when using https://TD-er.nl/ESPEasy/latest ? There are now so called 'CDC' builds that support the direct USB serial port on these boards. That might do the trick in flashing these units more easily. The button-press sequence as described is _**required**_ to get that serial port visible for upload in either Windows, Linux or Mac.

s0170071, 2023-06-11:
Yes, I did. Using Chrome. 
[image]

TD-er, 2023-06-11:
The S2 does only have a software emulated USB serial port.
This means it is simply not possible to toggle the unit into flash mode without user intervention.

So what you must do is keep the "Boot" button pressed and press reset.
In Windows device manager you will then see it appear using a completely different device (even though it might get the same com port nr assigned) and another "other device" called "ESP32-S2".
After this you can use the web flasher.

esptool.py might be able to toggle the unit into flash mode and flash it (hardly ever works predictable).
The reason why this can work is that it doesn't need to ask user permission to access the serial port, which is needed for the web flasher.

For this exact reason, I also configured ESP32-S3 builds to use the HWCDC port instead of USBCDC (software emulated), as HWCDC will remain accessible when toggling the device into flash mode.

N.B. for the initial first time flash you may still need to press boot/reset on the ESP32-C3 and -S3 to flash an ESPEasy build with CDC support (right now only the /latest builds on my web flasher page)

s0170071, 2023-06-11:
Thanks for the insight. This makes the S2 pretty inconvenient. I am about to toss them into the bin anyway. I had one working and after re-plugging it is not. Spend hours on it already and it feels so inmature.

closed 2023-06-11 by s0170071

TD-er, 2023-06-11:
Those purple S2 boards are indeed "a bit flaky" (well I would use other words, but let's keept it a bit more decent here)

Especially these purple boards have really poor antenna designs and it is really a matter of luck whether they will show a stable connection or not.

The SeeedStudio C3 boards are really nice as they have their own external antenna and are really small too.
They are my new favorite when it comes to 'small form factor'

I also like the S3 boards, but mainly when using original Espressif modules.
Not the ones with the ESP chip directly soldered to a PCB as those often have poor designs. However I guess ones with an IPEX connector (if they exist) would probably work just fine.

Just keep an eye on how the flash is linked to the S3 chip as that's kinda important or else it will bootloop.
Either octal SPI or quad SPI for flash and also for PSRAM, so that makes "only" 4 different combinations to pick from :)
You gotta love those idiotic design choices...
See: https://espeasy.readthedocs.io/en/latest/ESPEasy/ESPchips.html#quad-octal-spi-mode

Mice30, 2024-08-13:
> Just wanted to give a heads up for users of the purple S2 mini v1.0.0 board.
> 
> ![grafik](https://private-user-images.githubusercontent.com/33855380/244944143-f9447769-d17a-4215-9b90-1522ed59a063.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM1NDMzMDQsIm5iZiI6MTcyMzU0MzAwNCwicGF0aCI6Ii8zMzg1NTM4MC8yNDQ5NDQxNDMtZjk0NDc3NjktZDE3YS00MjE1LTliOTAtMTUyMmVkNTlhMDYzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODEzVDA5NTY0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdkNjZiZmRiMDAzNDJjOTI1OTBiNTFhNjFmNTU1YmQ1NWIwNWQ0ZGMyZmQ3ZTRmNjU4ODk2N2Q2NWZkYTg0YzImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.q3rkv66J6dcX5C-6agQ3I5YKCb1ERxzL9OZJvmUn5Os)
> 
> These were giving me a very hard time programming. I finally succeeded using
> 
>     * windows and the latest esp [tool v3.9.4 ](https://www.espressif.com/en/support/download/other-tools?keys=&field_type_tid%5B%5D=624). Use the S32 bin file with "factory" in the file name. Start address is 0. Press both buttons, release reset first, then the other. Only then the port will show up.
> 
>     * on Linux (mint) I managed to use the Arduino IDE but only if you quickly and timely reset the board after you hit the upload button.
> 
> 
> What was not working (but did with other boards) was TD-ERs upload webpage. Still nice job, @TD-er :-) Neither did the ESP Tool on linux, any other prior verions of the ESP Tool on Windows or the Arduino IDE on Windows.
> 
> I like the form factor of these S2 mini boards but I would not recommend buying them. Espressif seems to get lost in their various chip versions lately.

Can you share exact file and command line witch you have used, pls:). I am struggling...

tonhuisman, 2024-08-13:
2 things are important:
- try to avoid these boards
- to go into flash-mode: hold the `boot` button pressed, then shortly press the `reset` button, while keeping `boot` pressed. Now you can start your favorite flash-tool.

For flashing an ESP32 you must use the matching proc3ssor model (ESP32s2 for these boards), Flash-memory size (most likely 4MB for these boards), and a file that ends in `.factory.bin`, as that includes the required boot partition. The starting address for writing must be set to 0x0.

TD-er, 2024-08-13:
For some S2 boards you may also need to reconnect the USB cable while holding the boot button pressed and keep it pressed while starting the flash as the flasher may try to reboot the ESP and thus reset the USB mode.

The S2 does actually emulate a different USB device when it is in flash mode compared to normal mode.

If this doesn't work, you can also try plugging the ESP board in a different USB port as your OS may associate the complete path to the USB device with its installed driver. (path as in which port on which USB hub is needed to reach the device)