letscontrolit/ESPEasy #4701: OLED display weirdness

Budman1758, 2023-06-12:
I have several of these displays and am having the same issue with all of them. [Link to them.](https://www.diymore.cc/collections/hot-sale/products/2-42-inch-12864-oled-display-module-iic-i2c-spi-serial-for-arduino-c51-stm32-green-white-blue-yellow)

I have done the mods needed to make them work with I2C and the issue is they only work randomly if at all. When they do work, they work fine but most of the time they do not "lite up" especially after a reboot. Sometimes the display is scrambled, sometimes not. I am reasonably sure they are connected correctly because if not connected they way I have them they never work at all.

Connections to the RES and DC pins are connected to 3.3v. Won't work at all without those. CS pin seems to have no effect at all.

Looks to me like the actual support for I2C with these displays is marginal but I'm no expert. I have another I2C only display on the boat from China but it would be nice to be able to use these. I have 5 of the friggin things.

I have the log level set to debug more and don't see anything at all in them that looks suspect.

How hard would it be to support these kind of displays with SPI in ESPEasy? It's actually kind of difficult to find I2C only. Most are like this kind in question.

tonhuisman, 2023-06-12:
I'm quite sure that most of the issues are caused by incorrect initialization of the display controller, as ESPEasy supports the SSD1306 (and SH1106), not the SSD1309, as that needs a slightly different initialization. It's kind of a 'close but no cigar' situation.

What also might help is adding a capacitor/resistor combo to keep the RES pin low for a couple of milliseconds after power-on.

I've been thinking about adding support for these displays because of the larger 2.4" size over the regular SSD1306 displays (largest of those I've seen is 1.3" AFAIR, and my eyes are complaining about that 🙄), but I don't own that type of board yet, and the increasing amount of pending ESPEasy projects on my TODO list has stopped me until now... 😉

Budman1758, 2023-06-12:
I didn't realize they were an unsupported chip.  Oops!  I have several other older displays that are this size that are I2C only and have never had any problems with them. Not sure now what chip they are.....

As for the RES pin how does that hook up? Resistor and cap inline to ground on the pin connected to VIN?

TD-er, 2023-06-12:
I know the SSD1306 and the SH1106 do have some internal DC/DC circuit to generate the required internal negative voltage for these displays.
This also controls the intensity of the pixels.

For some displays this was quite tricky to setup as you would otherwise hear some coil whine and some even don't 'start' or show a flickering image.
So maybe these displays you're having also require some finetuning to set the correct registers for this contrast setting.
I know I have been tweaking those settings for a while listening very carefully, but that was years ago (around 2018)

Budman1758, 2023-06-12:
I can have one of these shipped to you guys if it would help with the support.  Or send you the funds to purchase one. Let me know.

Budman1758, 2023-06-13:
These were about half of current prices 2 or 3 years ago. No idea why they have gone up so much.

TD-er, 2023-06-13:
Maybe they have been featured in some kind of trending project?

Lots of parts have become much more expensive over the past 2 - 3 years, like BME280 and BN220 GPS.
Those were 3x - 4x the price they once were.
Right now I do see the BME280 at least being sold again for about 2,50 euro, which is close to what it once was.
And also new models of GPS modules from the same brand as those BN220 are now being sold again for around 9 euro.

So perhaps the price of these displays will also get to normal prices again.
Or maybe the controller used in these is no longer being manufactured and then you either need to adapt your software (and perhaps enclosures too) or pay the higher price.

Budman1758, 2023-06-22:
Just thought I would update this 
I received the other display I ordered and hooked it up. It works fine. No issues I can see so far. It is I2C only. Chip is a SSD1309 like the one I was using above so it must just be the difference between I2C and SPI initialization code. Might not be too difficult to sort out. Or maybe add SPI support for displays like this so they don't have to be modified to work.?  I still do have a few of them....  Hint hint nudge nudge.   :)

 [Display link](https://www.ebay.com/itm/295243268443?var=593440656504).