letscontrolit/ESPEasy #4702: No DNS entries when using Ethernet

kuppe234, 2023-06-13:
Hello,
I am using ESPEasy in conjunction with Ethernet. Wifi is not in use, there is no connection.

With the DHCP request, the DNS servers are also transmitted (visible in the log), but the variables (dns0_cache and dns1_cache) are not set and therefore 0.0.0.0. This is due to an error in "src/src/DataStructs/EthernetEventData.cpp". "WiFi" must be changed to "ETH" there. 

See "git diff":
```
diff --git a/src/src/DataStructs/EthernetEventData.cpp b/src/src/DataStructs/EthernetEventData.cpp
index 01a8e25b8..592177d92 100644
--- a/src/src/DataStructs/EthernetEventData.cpp
+++ b/src/src/DataStructs/EthernetEventData.cpp
@@ -5,7 +5,7 @@
 #include "../ESPEasyCore/ESPEasy_Log.h"
 #include "../Helpers/Networking.h"
 
-#include <WiFi.h>
+#include <ETH.h>
 
 // Bit numbers for Eth status
 #define ESPEASY_ETH_CONNECTED               0
@@ -99,11 +99,11 @@ void EthernetEventData_t::setEthConnected() {
 bool EthernetEventData_t::setEthServicesInitialized() {
   if (!unprocessedEthEvents() && !EthServicesInitialized()) {
     if (EthGotIP() && EthConnected()) {
-      if (valid_DNS_address(WiFi.dnsIP(0))) {
-        dns0_cache = WiFi.dnsIP(0);
+      if (valid_DNS_address(ETH.dnsIP(0))) {
+        dns0_cache = ETH.dnsIP(0);
       }
-      if (valid_DNS_address(WiFi.dnsIP(1))) {
-        dns1_cache = WiFi.dnsIP(1);
+      if (valid_DNS_address(ETH.dnsIP(1))) {
+        dns1_cache = ETH.dnsIP(1);
       }
 
       #ifndef BUILD_NO_DEBUG
```

Greetings, kuppe.

tonhuisman, 2023-06-13:
What ESPEasy release are you running?
Have you tried to install the latest available from http://td-er.nl/espeasy/latest/ ? (⚠️ be careful when connecting Ethernet and USB at the same time to your ESP, especially when using PoE! Better unplug the Ethernet cable when connecting to USB ⚠️)

TD-er, 2023-06-13:
Made a PR for this:
https://github.com/letscontrolit/ESPEasy/pull/4703

kuppe234, 2023-06-13:
I don't have wifi here at all. Therefore, no DNS entries can be configured via Wifi DHCP.

closed 2023-06-14 by TD-er

kuppe234, 2023-06-14:
1 second after receiving the DHCP data (with correct DNS entry) both DNS variables were set to 0.0.0.0.
Thanks for the change in git, now everything works perfectly.

TD-er, 2023-06-14:
Yep, the DNS fields will be cleared when WiFi.mode() changes.
Very frustrating.

I'm thinking about adding very strict networking modes in ESPEasy, so WiFi doesn't even need to be started when a user needs Ethernet only.

Right now I try to make it as dynamic as possible, so if the Ethernet plug is removed it will start WiFi etc.
But since it does take a while to setup Ethernet, WiFi is already started now and that's causing issues like these.

But that's a major overhaul of ESPEasy code, so I'm holding that off a bit as it is a good guarantee for lots of other side issues when doing such a major change and I need to free up quite a lot of time to spend on it continuously.

PR #4703 (2023-06-14): [ETH] No DNS entries when using Ethernet (#4702)
Fixes: #4702
commit dbeb1ad6 (2023-06-14): [ETH] No DNS entries when using Ethernet (#4702)

Fixes: #4702