letscontrolit/ESPEasy #4706: Such a problem occurs when using platformio to compile firmware

qiudecai, 2023-06-16:
I recently encountered such a problem when using platformio to compile firmware, I don't know where I went wrong, and I don't know how to solve it

Processing normal_ESP32_4M316k (board: esp32_4M; platform: C:\Users\qdc\.platformio\platforms\espressif32; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
 Compile time defines 
 File System:      SPIFFS
 PROGNAME:         ESP_Easy_mega_20230616_normal_ESP32_4M316k
 BOARD_NAME:       Espressif Generic ESP32 4M Flash ESPEasy 1810k Code/OTA 316k FS
 BUILD_PLATFORM:   Windows-10-10.0.19044-SP0
 GIT_HEAD:         pygit2_not_installed
 CDN_URL:          https://cdn.jsdelivr.net/gh/letscontrolit/ESPEasy/static/
 BUILD_VERSION:    20502
 BUILD_UNIXTIME:   1686924875
 BUILD_RFC1123:    Fri, 16 Jun 2023 14:14:35 GMT
 ------------------------------- 
 Flash configuration 
 --flash-size:   4MB
 --flash-freq:   40000000L
 --flash-mode:   dio
  memory_type:   dio_qspi
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32_4M.html
PLATFORM: Espressif 32 (6.3.1) > Espressif Generic ESP32 4M Flash ESPEasy 1810k Code/OTA 316k FS
HARDWARE: ESP32 240MHz, 320KB RAM, 1.81MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Converting ESPEasy.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 97 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit HMC5883 Unified @ 1.2.0
|-- Adafruit Motor Shield V2 Library @ 1.0.4
|-- Adafruit NeoPixel @ 1.10.6
|-- Adafruit Unified Sensor @ 1.0.2
|-- AM2320
|-- ArduinoJson @ 6.19.4
|-- AS_BH1750
|-- ESPeasySerial @ 2.0.16
|-- GPIO_DirectAccess
|-- HLW8012 @ 1.1.1
|-- SerialSensors @ 0.0.1
|-- PZEM-004T-v30 @ 1.0.0
|-- MechInputs @ 0.0.1
|-- Regexp @ 0.1.0
|-- SDM @ 2.0.0
|-- SPI @ 2.0.0
|-- TinyGPSPlus @ 1.0.2
|-- WakeOnLan @ 1.1.6
|-- Blynk @ 1.0.1
|-- Ethernet @ 2.0.0
|-- SD @ 2.0.0
|-- ServoESP32 @ 1.0.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- RTClib @ 1.13.0
|-- Wire @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- Update @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|-- LiquidCrystal_I2C
|-- ESP8266_SSD1306 @ 3.2.7
|-- Adafruit TCS34725 @ 1.0.1
|-- CHT16K33 LED Matrix Library @ 0.0.1
|-- Adafruit MPR121 @ 1.0.0
|-- SparkFun APDS9960 RGB and Gesture Sensor @ 1.4.2
|-- Adafruit SGP30 Sensor @ 1.0.5
|-- XPT2046_Touchscreen
|-- MD_MAX72XX @ 3.3.0
|-- MD_Parola @ 3.5.6
|-- Adafruit BME680 Library @ 2.0.2
|-- Adafruit SI1145 Library @ 1.1.1
|-- MFRC522
|-- SparkFun_AS7265x
|-- FrogmoreScd30 @ 1.0
|-- IthoCC1101 @ 1.0.0
|-- Multi Channel Relay Arduino Library @ 1.1.0
|-- ShiftRegister74HC595_NonTemplate @ 1.3.1
|-- Adafruit NeoMatrix @ 1.3.0
|-- RAK12019_LTR390_UV_Light @ 1.0.1
|-- SparkFun SCD4x Arduino Library @ 1.0.4
|-- I2C AXP192 Power management @ 1.0.4
|-- htcw_ip5306 @ 0.1.0
|-- Adafruit seesaw Library @ 1.6.3
|-- NewPing @ 1.9.6
|-- Adafruit TSL2591 Library @ 1.0.2
|-- ccronexpr @ 1.0.0
|-- Adafruit GFX Library @ 1.11.3
|-- Adafruit ILI9341 ESPEasy @ 1.5.10
|-- LOLIN_EPD @ 1.0.0
|-- VL53L0X @ 1.3.0
|-- SparkFun VL53L1X 4m Laser Distance Sensor @ 1.2.9
|-- SparkFun MAX1704x Fuel Gauge Arduino Library @ 1.0.4
|-- Adafruit_ST77xx @ 1.7.3
|-- ITG3205 @ 1.0
|-- SparkFun ADXL345 Arduino Library @ 1.0.0
|-- NeoPixelBus @ 2.6.9
|-- Adafruit PCD8544 Nokia 5110 LCD library @ 2.0.1
|-- PubSubClient @ 2.7
|-- RN2xx3 Arduino Library @ 1.0.2
|-- I2Cdevlib-Core
|-- CircularBuffer @ 1.3.3
|-- DNSServer @ 2.0.0
|-- ESP32HTTPUpdateServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- WebServer @ 2.0.0
|-- LittleFS @ 2.0.0
|-- SPIFFS @ 2.0.0
TypeError: can only concatenate deque (not "list") to deque:
  File "C:\Users\qdc\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 174:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\qdc\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\qdc\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\qdc\.platformio\platforms\espressif32\builder\main.py", line 312:
    target_elf = env.BuildProgram()
  File "C:\Users\qdc\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\qdc\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 61:
    env.ProcessProjectDeps()
  File "C:\Users\qdc\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\qdc\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 163:
    plb.env.BuildSources(
  File "C:\Users\qdc\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\qdc\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 352:
    nodes = env.CollectBuildFiles(variant_dir, src_dir, src_filter)
  File "C:\Users\qdc\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\qdc\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 298:
    new_node = callback(new_node)
  File "E:\ESP8266\ESPEasy\ESPEasy-mega-20230508\tools\pio\generate-compiletime-defines.py", line 107:
    CPPDEFINES=env["CPPDEFINES"]
=============================================== [FAILED] Took 57.04 seconds ===============================================

Environment          Status    Duration
-------------------  --------  ------------
normal_ESP32_4M316k  FAILED    00:00:57.042
========================================== 1 failed, 0 succeeded in 00:00:57.042 ==========================================

tonhuisman, 2023-06-16:
You seem to have downloaded the source .zip from the 20230508 release of ESPEasy.
Since then, PlatformIO has changed some internals (keyword: scons) that cause the python scripts we use, to break on the build. These issues have been fixed on the `mega` branch, so if you use that up-to-date source your build should succeed.

We also have a [Developers guide](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html), that includes instructions for creating a Custom build.
For regular builds like `normal`, there is no reason to create your own builds, they are all included in the downloadable .zip files.

qiudecai, 2023-06-17:
I want to compile the previous project firmware, what do I need to do?

tonhuisman, 2023-06-17:
> I want to compile the previous project firmware, what do I need to do?

While that's not a desirable or advisable option, you _should_ be able to do that by downgrading PlatformIO to v6.1.6 or older, as from PlatformIO v6.1.7 the SCons package has been updated to a version that caused these issues.

Solid advice: Get the latest `mega` sources, as that will soon be the next release version of ESPEasy. And like any OSS project: The latest is the greatest (and the only supported release...).

Just being curious: What is the reasoning to try and build an older release?

qiudecai, 2023-06-18:
Thank you! The problem is solved and I tried modifying the tools\pio\generate-compiletime-defines.py file

qiudecai, 2023-06-18:
Build an old release because I've been using it before

chemmex, 2023-06-19:
> Just being curious: What is the reasoning to try and build an older release?

I stick to older releases for ESP8266 because latest ones do not start AP in case of WiFi misconfiguration or network absence, thus making re-provisioning impossible.

chemmex, 2023-06-29:
> Thank you! The problem is solved and I tried modifying the tools\pio\generate-compiletime-defines.py file

Can you share your modifications please?

tonhuisman, 2023-06-29:
> Can you share your modifications please?

This [commit](https://github.com/letscontrolit/ESPEasy/pull/4656/commits/a0dca6ca05a9aed7ee05cb6cec034345a07a86d3) has the relevant changes (quite minor, but crucial 😅)

TD-er, 2023-06-29:
> > Can you share your modifications please?
> 
> This [commit](https://github.com/letscontrolit/ESPEasy/pull/4656/commits/a0dca6ca05a9aed7ee05cb6cec034345a07a86d3) has the relevant changes (quite minor, but crucial 😅)

Not only 'crucial', but also took me > 10h to find and fix...

qiudecai, 2023-06-30:
> 

This is after I modified

qiudecai, 2023-06-30:
I replaced it from the latest mega download

chemmex, 2023-06-30:
Oh, thanks. One never knows how much effort hides behind a small bugfix or update

tonhuisman, 2023-09-09:
This issue can be closed, as it is solved.
The issue mentioned by chemmex is already registered as a separate issue.

closed 2023-09-09 by TD-er