letscontrolit/ESPEasy #4709: MQTT - Send value only if it's different from the previous one

iz8mbw, 2023-06-20:
Hello.
Just a question or if what I'm asking it's not implemented for me is a "nice to have" feature.

Think to this scenario: I have ESPEasy with various "climate" sensors and ESPEasy send via MQTT the values, for what I know currently the frequency to send the values via MQTT is the same to "Interval" value (sensor polling) configured in the Device page, so if the sensor polling interval is 30 seconds the value sent to MQTT is every 30 seconds.
So for example I have an "Interval" value set to 30 seconds, ESPEasy will send the sensor value to MQTT every 30 seconds.

Is it possible to tell to ESPEasy to send via MQTT the "new" value only if it's different from the previous one?

In this way if I poll the sensor every 30 seconds but the new value returned from that sensor happens after 90 seconds ESPEasy will send the new value at the 90th second.

Thanks!

TD-er, 2023-06-20:
Sure this is possible.

For example you need to watch the task called "foo" and its task value is "bar".

Just add a dummy task which is linked to MQTT and not the task that does interact with the sensor.

Maybe you also want to use the taskname and taskvaluename in the MQTT topic. If so, then you may want to rename the original task and use those names for the dummy task.

Rename task "foo" to "foosensor".
Name the new dummy task "foo" and its taskvalue "bar".

"foosensor" will generate an event on each reading: `foosensor#bar=1234`

In the rules you make a block like this:

```
on foosensor#bar do
  if [var#1]!=%eventvalue1%
    let,1,%eventvalue1%
    taskvalueset,foo,bar,%eventvalue1%
    taskrun,foo
  endif
endon
```

It will use a variable (`[var#1]`) to keep track of the last value. If it differs, the variable is updated and set the value taken from the event into the dummy task called 'foo' and then call TaskRun, which will send the value to the linked controller.

TD-er, 2023-06-20:
Problem is that not all controllers act the same.
For example Domoticz controllers send all task values at once. (and need to send them at once)
So that makes it already rather complex as you need to decide per controller what would be considered "the same value".

The next request would be to add some hysteresis to consider something the same value.
Or "only send when it quickly rises" (first derivative > X)

I do understand this may seem a bit 'complex', but it will quickly become more complex to add such filtering requirements.

What would be a better solution is to actually  introduce a new concept in ESPEasy to insert a filter inbetween the task and the controller.
That's however quite a lot of work, which is what I'm working on. But that will take a while as I'm still trying out several concepts before I will release that.
A very preliminary step is what now can be used via the "stats" option per task, which allows you to take averages, compute standard deviation, etc. Still only really usable as filter via rules, but in the end it will be used as some new concept to insert between a task and a controller. (or actually as a 'callback' from the controller to decide whether or not to forward the data)

But that's far from done yet and may take quite a while to implement.
The reason I'm already telling about it is to let you know that I will not implement it like you requested it, but in a way it will be added somewhere in the future.  (several months to maybe a year from now)

iz8mbw, 2023-06-21:
Hi.
Also if Domoticz controllers send all task values at once and for example we have 3 values to send and of these 3 values only one is changed I expect that in this case all the 3 values will be sent and only one have a different value.
Instead, if all the 3 values are the same then do not send any (new) MQTT message, I mean if at least one value is changed and "the array" is made by 3 values it's mandatory to send (anyway) all the values (that will contains 1 new and the other 2 the same of before).

In my case I use "Home Assistant (openHAB MQTT)" and I have a single value per single field.

TD-er, 2023-06-21:
I know what you mean and understand your request.
But from a UI and/or settings point of view this will become either complex to operate or complex code to manage.
Both aspects will trigger allergic reactions to me :)

So as I tried to explain, eventually there will be a similar feature, but it does take quite a lot of changes to do it right.
Meaning right now you can only limit the message rate via the rules like I suggested.

To do it the right way, a new concept within ESPEasy needs to be added and right now I am still trying several options to see which one will be the most intuitive to use.
It will be in the form of a 'filter', but I'm not yet sure whether this should be a property of a task value or some 'meta controller'.
The idea of the latter is that you could create chains of "controllers" or "filters" or whatever it will eventually be called.

One of the problems I also want to solve with these is to have some kind of way to sync tasks so you can act on a trigger to get samples from several sensors as close as possible at the same time.
But not all sensors need the same amount of time to take a sample.

The filtering idea you're having is also part of this as it needs to decide what values to forward to a controller and what values should be blocked... or further processed. (e.g. averaging, filtering, etc.)

In other words, eventually there will be a similar feature as you suggested, but not right now.

iz8mbw, 2023-06-21:
Thanks!
So maybe better to change the label in Github from "Type: Question" to "Type: Enhancement" otherwise it will never be implemented. 😊

iz8mbw, 2023-06-22:
> on foosensor#bar do
>   if [var#1]!=%eventvalue1%
>     let,1,%eventvalue1%
>     taskvalueset,foo,bar,%eventvalue1%
>     taskrun,foo
>   endif
> endon
> ```
> 
> It will use a variable (`[var#1]`) to keep track of the last value. If it differs, the variable is updated and set the value taken from the event into the dummy task called 'foo' and then call TaskRun, which will send the value to the linked controller.
> 
> N.B. questions like these are best asked at the forum, as GitHub issues are about bugs and feature requests.

I tried but I cannot reach the goal, [I wrote a post on the Forum](https://www.letscontrolit.com/forum/viewtopic.php?t=9709) about the Rules. Thanks.