letscontrolit/ESPEasy #4710: Sending Tag presence double times to FHEM

rsp-git, 2023-06-20:
Hello,
i tried a new installation of ESP Easy to connect a RFID Reader with WemosD1 to FHEM over WiFi.

If i put the Tag in front of the Reader it Beeps and the Wemos sends the ID - like it should be.
After a while it sends the ID again - this should not be.
The Time can be 30sec and also 5 minutes.

Actual i have this config:
  Controller:
    FHEM HTTP
    Allow Expire on
    De-Duplicate on
  Device:
    RFID - Wiegand 34 bit
    auto Tag Remove
    Single Event with all values
    
   I Played arround but nothing changed.

This is a Log from FHEM:
2023-06-01 15:44:47 ESPEasy ESPEasy_RFID Tag: 1234
2023-06-01 15:44:47 ESPEasy ESPEasy_RFID presence: present
2023-06-01 15:44:47 ESPEasy ESPEasy_RFID Tag: 1234
2023-06-01 15:48:40 ESPEasy ESPEasy_RFID presence: present
2023-06-01 15:48:41 ESPEasy ESPEasy_RFID Tag: 1234

How can i define if its from this module, from FHEM Connector or from Fhem?

So maybe you can help me.

rsp-git, 2023-06-20:
No, there are no Rules defined.
Only one Device and one Controller.

TD-er, 2023-06-20:
Is it possible the controller may not receive an acknowledgement and does retry sending it again?
You can test this by setting the number of retries of that controller to 0.
See: https://espeasy.readthedocs.io/en/latest/Controller/_Controller.html#controller-parameters

rsp-git, 2023-06-21:
Hello,
i've tried a little bit arround and this Parameter can only be set to min. 1. But this does nothing change.
BR

tonhuisman, 2023-06-24:
Setting the `Automatic tag removal` option enabled, and the value for that to 0, should at least not report the last scanned tag to FHEM.

Have you checked if the unit is rebooting unexpectedly, possibly because of power supply issues? That, combined with Automatic tag removal disabled, will cause the last tag value to be re-sent when the plugin is started after the reboot.