letscontrolit/ESPEasy #4711: P093 - Mitsubishi Heat Pump Add Remote Temperature

jfmennedy, 2023-06-20:
Hello,

might it be possible to add setRemoteTemperature to Plugin 093 as implemented at SwiCago https://github.com/SwiCago/HeatPump/blob/e9fc858374b340a54b4f31fac6423af61b7b77f2/src/HeatPump.cpp#L273

That would be nice and improve the operating of internal devices as the internal temperature sensor is not as reliable to match the real room temperature

Thx, Jan

crnjan, 2023-06-23:
Shouldn't be too hard to add, will check when there is a bit of (free?) time ...

jfmennedy, 2023-09-15:
Hi, is there any news on this ?

jfmennedy, 2023-09-19:
Well I tried it myself, and created a PR.. Might be reviewed before release...

jfmennedy, 2023-09-22:
someone can help here?:

https://github.com/letscontrolit/ESPEasy/pull/4813#discussion_r1334789701

PR #4813 (2023-10-22): P093 Mitsubishi HeatPump
Add Support for SetRemoteTemperature
Resolves P093 - Mitsubishi Heat Pump Add Remote Temperature 
Fixes: #4711
closed 2023-10-22 by TD-er