letscontrolit/ESPEasy #4714: TypeError: can only concatenate deque (not "list") to deque

JoostDkr, 2023-06-23:
Until recently I was able to compile EspEasy with VS-studio/Platform-IO on Windows.
Suddenly I have compile-errors.
I am using ESPEasy-mega-20230508 but older version gives the same error.
I can't figure out what the cause is. Who can help me?

I get the following error:

**TypeError: can only concatenate deque (not "list") to deque:**

   File "C:\Users\JDekker\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 174:
     env.SConscript("$BUILD_SCRIPT")
   File "C:\Users\JDekker\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
     return _SConscript(self.fs, *files, **subst_kw)
   File "C:\Users\JDekker\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
     exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
   File "C:\Users\JDekker\.platformio\platforms\espressif32@src-0b030f7ca60e6bce2d9c413ec210f70f\builder\main.py", line 292:
     target_elf = env.BuildProgram()
   File "C:\Users\JDekker\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
     return self.method(*nargs, **kwargs)
   File "C:\Users\JDekker\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 61:
     env.ProcessProjectDeps()
   File "C:\Users\JDekker\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
     return self.method(*nargs, **kwargs)
   File "C:\Users\JDekker\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 163:
     plb.env.BuildSources(
   File "C:\Users\JDekker\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
     return self.method(*nargs, **kwargs)
   File "C:\Users\JDekker\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 352:
     nodes = env.CollectBuildFiles(variant_dir, src_dir, src_filter)
   File "C:\Users\JDekker\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
     return self.method(*nargs, **kwargs)
   File "C:\Users\JDekker\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 298:
     new_node = callback(new_node)
   File "C:\Users\JDekker\Documents\PlatformIO\Projects\ESPEasy-mega-20221001+P141\tools\pio\generate-compiletime-defines.py", line 101:
     CPPDEFINES=env["CPPDEFINES"]

_**Errors are in file "_generate-compiletime-defines.py_**_"
**The first error is on the 1e line "Import ("env")". where "Import" is not defined.**

Who can help me?

TD-er, 2023-06-23:
There has been an update in PlatformIO which breaks the ESPEasy build.

On older versions of the source code you need platformio 6.1.6 (not 100% sure if it is <6.1.6 or <=6.1.6)
In the latest code you can use the latest version of platformio.

tonhuisman, 2023-06-23:
See also #4706

JoostDkr, 2023-06-23:
Thanks for answering.
I'm using:
1/  the "PlatformIO Core" version 6.1.7
2/ the ESPEasy-mega-20230508

But unfortunately I still have the compile errors in file
.\tools\pio\generate-compiletime-defines.py. The compiler
1/ " can not find pygit2" and
2/ give a error on in line 1: "Import(env)" : "Import" is not defined.

Also older versions of "PlatformIO Core" give's these errors.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusvrij.www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#m_-5122955281645555837_m_-4824535440266435855_DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

TD-er, 2023-06-23:
> 1/  the "PlatformIO Core" version 6.1.7
> 2/ the ESPEasy-mega-20230508

That's the combination which will NOT work.

Why are you trying to use the older source code?
Downloading the latest code would probably be a lot easier as for me it is guessing what will be the problem here.

You can open the CLI in VS code and run `pip install -r requirements.txt`
This will install pygit2.

However the older version of the code does probably not have a limit on the platformio version (in requirements.txt) as it was unclear to me back then that PlatformIO would introduce a breaking change.

JoostDkr, 2023-06-23:
Running " pip install -r requirements.txt" did not help, unfortunately.
I do not understand what you mean by older version.
I'm assuming I'm using the latest versions:
1/ ESPEasy-mega-20230508
2/ PlatformIO Core, version 6.1.7
3/ VS 1.79.2

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusvrij.www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

tonhuisman, 2023-06-23:
> 1/ ESPEasy-mega-20230508
> 2/ PlatformIO Core, version 6.1.7

Shortly after the ESPEasy release 20230508 was released, PlatformIO updated to 6.1.7, that has a *breaking* change in one of the Python libraries used, resulting in a build failure on one of the (core) Python scripts used in ESPEasy. This has been fixed shortly after that, but hasn't been included in a release yet.
The _only_ way to build that release of ESPEasy is to use an older release of PlatformIO (**6.1.6 or older**) to build ESPEasy, OR use the latest `mega` sources of ESPEasy to build a working version without that error popping up.

When building a non-standard veresion of ESPEasy it is always advised to the latest `mega` sources to start with, _not_ a downloaded zip of the sources.
As always in OSS (Open-source software): The latest is the greatest 😃

TD-er, 2023-06-23:
Just pushed a new tag to build a new release...
So the latest and greatest sources are of today :)  (was planning to do a new build anyway)

Just download the latest source code: https://github.com/letscontrolit/ESPEasy/archive/refs/heads/mega.zip

JoostDkr, 2023-06-24:
Yeah........,
Espeasy Mega Now compiles without errors!
Thanx.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusvrij.www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2023-06-24 by JoostDkr