letscontrolit/ESPEasy #4715: P064: "Gesture - APDS9960" and P012: "LCD2004" are not possible to enable.

JoostDkr, 2023-06-24:
It is not possible for me to enable these two plugins.
I can check "Enabled" to "On" but after "Submit" they are unchecked again.

I using the latest EspEasy-Mega.

tonhuisman, 2023-06-24:
Do you have some (error) message in the log?
Does the I2C scanner show the addresses? (and depending on the build used, also the device/task names)

We have added in Tools/Advanced a setting `Check I2C devices when enabled` that's checked by default. You can uncheck that and see if that solves your issue.

JoostDkr, 2023-06-24:
The ESP is not connected to hardware. I'll first preparing the ESP-software for a new application.
"setting Check I2C devices when enabled" helps for the LCD2004 not for the APDS9960.

In the logfile: there is tis entry: 862917: APDS : Error during APDS-9960 init!

I'll try again after the ESP is connected to application-hardware.

tonhuisman, 2023-06-24:
> The ESP is not connected to hardware. I'll first preparing the ESP-software for a new application.

Without the hardware connected it is rather useless to enable the plugin, as it will immediately try to initialize the hardware, and that will of course fail. With that setting you now disabled, a check on task-startup (PLUGIN_INIT) has been introduced to verify if the configured I2C device can be found, and when that fails the plugin will be disabled (that's what happened for the LCD2004). And also, plugins will be disabled if they fail to properly initialize, like the APDS9960 shows, that's independent of the device being I2C or something else.

You should best re-enable the `Check I2C devices when enabled` setting, as that's not the real cause of this behavior.

TD-er, 2023-06-24:
The reason this protection, to disable tasks which cannot access the sensor, was added to make sure the ESPEasy node remains accessible.
Problem with non connected sensors is that it can freeze due to lots of timeouts.

If you need to prepare an image for other units, you can also call "taskenable,<taskname>" in the rules in a block `on system#boot do` block.
The protection will then disable the tasks again if the sensor isn't connected, but will still try to enable it on each boot.

JoostDkr, 2023-06-24:
thanks for all  the information.
I added the hardware to the ESP and everything is working OK

closed 2023-06-24 by JoostDkr