letscontrolit/ESPEasy #4721: [Idea/feature request] Local CDN - Option to store required scripts on SD bypassing downloading from CDN

305651, 2023-06-28:
Hello, I have an idea to make possible to store scripts and other web-related files in a folder on SD card, when build is compiled with SD support. It would allow to use web interface on networks without Internet access and also will work-around issues like #4717. Ideally the web interface should have an option to download all required files once and write them to SD.

Pros:
Independent of CDN if it goes offline
Allow users to further customize scripts
SD Card support is already included in ESPEasy
Workaround some issues like #4717 

Cons:
May require preparing custom builds with SD support as it is not included by default
Unsure if web interface can access files on SD on-the-fly, maybe we can store them on different medium ( external EEPROM?)

This is only an idea, feel free to close this issue if it is out of the scope or too hard to implement. Thank You.

TD-er, 2023-06-28:
If SD card support is enabled in the build, this feature already will work.

However right now I don't think we do have builds with SD-card support enabled in the official builds (maybe the "max" builds???)

tonhuisman, 2023-06-29:
> (maybe the "max" builds???)

Yes, all MAX builds do have SD-card support enabled, and some of the MAX builds even have 8MB of local storage available on flash, and that can also be used to store these files.
But for the regular 4MB ESP32 builds there is only 316kB local storage and on 1MB ESP8266 builds only 128kB, and both use roughly half of that for storing the settings and rules files, so it might be challenging to upload the CDN-based files there.
The 4MB ESP8266 builds have 1MB available for local storage, so there it should be possible to upload at least some of these files, as the settings files on ESP8266 are somewhat smaller, as they only need to store settings for 12 tasks compared to 32 tasks for ESP32.

> Unsure if web interface can access files on SD on-the-fly,

Any file is first looked for on internal storage, and when not found and SD-card support is configured/enabled, the file is searched for on the SD-card. That's the standard behavior.
Theoretically it is possible to move all files from internal flash to SD-card, but I haven't tried that yet 😅 

TL;DR;
This feature request is already working as requested, except that SD-card support is not available is all builds (code-size limitations).

305651, 2023-06-29:
Thank you for reply, actually I was aware of saving scripts on internal storage but I didn't know that it is also true for SD. I think it can be closed now.

closed 2023-06-29 by 305651