letscontrolit/ESPEasy #4727: New device support GY-68 (BMP180)

almirus, 2023-07-09:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

Zero values

### Expected behavior

Non zero values

### Actual behavior

Temperature:0.00
Pressure:0.00

### Steps to reproduce

1. Connect **GY-68 (BMP180)** to ESP8266 via i2c (power + SDA: GPIO-4 (D2) SCL: GPIO-5 (D1))
2.  Add new device on Tab Devices: Environment - BMP085/180
3. 

### System configuration

Hardware:
ESP Board
--
ESP Chip ID: | 1358343 (0x14BA07)
ESP Chip Frequency: | 80 MHz
ESP Chip Model: | ESP8266
ESP Chip Cores: | 1
ESP Board Name: | Espressif Generic ESP8266 ESPEasy 4M Flash 1M FS
Storage
Flash Chip ID: | Vendor: 0x20 Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash Chip Speed: | 40 MHz
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DOUT
Flash Writes: | 5 daily / 5 boot
Sketch Size: | 944 kB (2124 kB free)
Max. OTA Sketch Size: | 1019 kB (1044464 bytes)
OTA possible: | true
OTA 2-step Needed: | false
SPIFFS Size: | 934 kB (844 kB free)

ESP Easy version: 
Build: [ESP_Easy_mega_20230623_climate_ESP8266_4M1M Jun 23 2023](https://github.com/letscontrolit/ESPEasy/releases)
Firmware
--
Build:⋄ | 20230623 - Mega
System Libraries:⋄ | ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | mega-20230623
Plugin Count:⋄ | 59 [Climate][No Debug Log]
Build Origin: | GitHub Actions
Build Time:⋄ | Jun 23 2023 21:19:58
Binary Filename:⋄ | ESP_Easy_mega_20230623_climate_ESP8266_4M1M
Build Platform:⋄ | Linux-5.15.0-1040-azure-x86_64-with-glibc2.35
Git HEAD:⋄ | HEAD_ec086a4

ESP Easy settings/screenshots: 
[image]

[image]

[image]

### Rules or log data

```
On System#Boot do    //When the ESP boots, do
  looptimerset,1,300  // Start loop timer 1, 300 sec interval
  timerSet,2,30 // Set timer 2 to put the sensor to sleep again
endon

On Rules#Timer=1 do
  timerSet,2,30 // Set timer 2 to put the sensor to sleep again
  pmsx003,wake
endon

On Rules#Timer=2 do
  pmsx003,sleep
endon
```

TD-er, 2023-07-09:
So in other words, the BMP180 plugin doesn't seem to work on your sensor?

Can you try a (much) older build of ESPEasy to see if the sensor is working?
This plugin (P006) has hardly been changed in the last 4 - 5 years, except for some generic changes in ESPEasy's core code to handle I2C devices.
So there might be some bug in those changes and therefore if an old build of ESPEasy is working on your sensor we at least know the sensor is working.

TD-er, 2023-07-09:
Ah nope, scratch what I said... the AHT10 sensor is causing issues here...
See the [warning in the documentation](https://espeasy.readthedocs.io/en/latest/Plugin/P105.html#p105-page)

closed 2023-07-09 by TD-er

almirus, 2023-07-09:
@TD-er After disconnecting the AHT10 sensor, GY-68 sensor started working.

[image]

TD-er, 2023-07-09:
Yep, that AHT10 sensor is really crappy.
Was just discussing with Ton that we should add some extra warnings in the web UI of that plugin, to help users not to fall for this trap again.