letscontrolit/ESPEasy #4731: MCP23017 Inputs not updating correct values except after reboot.

Humanoid-XXX, 2023-07-12:
### Checklist
- [x ] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [x] given a list of active plugins or controllers when applicable.
- [x] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x] Tested previous/other build (mention which one already tested)
tried an old build from 2022
- [ ] Tested on other node to make sure hardware isn't defective.
- [x] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request
About once every 2 hours MCP23017 input readings freeze up and no longer register actual changes.
I have the MCP23017 set to report the values to domoticz every 30 seconds.
When the issue occurs espeasy still reports the incorrect previous values every 30 seconds.
The SHT3x that shares the I2C bus always continues to function properly. 
Note that i2c speed is 10,000hz due to long sensor cable for sht3x. 10Meter
There were a few times where the values were -1 but that hasnt happened for a while. 
if the esp is rebooted via software, the values update to the correct values and the unit operates how it should for an hour or 2

### Expected behavior
When input changes, it should always register and if it freezes up, it should not continue reporting false values

### Actual behavior
It freezes up and reports false values occasionally

### Steps to reproduce
1. Install all plugins that I have and all hardware. 
2. Ensure MCP23017 sends data to domoticz every 30s
3. Set MCP23017 inputs to pullup mode with inputs providing gnd when closed.
4. watch as inputs freeze and no longer update when changed

Resetting fixes the issue temporarily

### System configuration

Hardware:
D1 Wemos on custom breakout board with AC-DC power supply
MCP23017
SHT31
 Gems Sensors 224504 water level sensor (Provides sink)

ESP Easy version: 

Firmware
--
Build:⋄ | 20230623 - Mega
System Libraries:⋄ | ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | mega-20230623
Plugin Count:⋄ | 80 [Normal][Collection][Collection_A][No Debug Log]
Build Origin: | GitHub Actions
Build Time:⋄ | Jun 23 2023 21:20:25
Binary Filename:⋄ | ESP_Easy_mega_20230623_collection_A_ESP8266_4M1M
Build Platform:⋄ | Linux-5.15.0-1040-azure-x86_64-with-glibc2.35
Git HEAD:⋄ | HEAD_ec086a4
System Status
Syslog Log Level: | Error
Serial Log Level: | None
Web Log Level: | None
Enable Serial Port Console: | true
Console Serial Port: | HW Serial0 RX:3 TX:1 @ 115200 baud
I2C Bus State: | OK
I2C bus cleared count: | 0
Network Services
Network Connected: | ✔
NTP Initialized: | ✔
ESP Board
ESP Chip ID: | 1655971 (0x1944A3)
ESP Chip Frequency: | 80 MHz
ESP Chip Model: | ESP8266
ESP Chip Cores: | 1
ESP Board Name: | Espressif Generic ESP8266 ESPEasy 4M Flash 1M FS
Storage
Flash Chip ID: | Vendor: 0xD8 Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash Chip Speed: | 40 MHz
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DOUT
Flash Writes: | 48 daily / 48 boot
Sketch Size: | 1006 kB (2064 kB free)
Max. OTA Sketch Size: | 1019 kB (1044464 bytes)
OTA possible: | true
OTA 2-step Needed: | false
SPIFFS Size: | 934 kB (843 kB free)

ESP Easy settings/screenshots: 
Frozen State: 
[image]
[image]
[image]

Working state immediately after a software reboot:
[image]
[image]
[image]

### Rules or log data

```
On System#Boot do
MCPmode 1,2
MCPgpioRange 2,8,1
MCPGpio,1,0
timerSet,1,120

endon

on WaterLevelLowLow#State!=1 do 
if HumidificationPumpStatus#State=1 
GPIO,15,0
endif
endon

on [Pump_Temp#Temp]>110 do 
if HumidificationPumpStatus#State=1 
GPIO,15,0
endif
endon

on HumidificationPumpStatus#State=1 do
if [WaterLevelLowLow#State]!=1 or [Pump_Temp#Temp]>110
GPIO,15,0
endif
endon

on HumidificationDemand=1 do   // http://192.168.13.12/control?cmd=event,HumidificationDemand=1

TaskValueSet,6,1,1
if [WaterLevelLowLow#State]=1 and [Pump_Temp#Temp]<110
GPIO,15,1
Mcpgpio 1,1
endif
endon

on HumidificationDemand=0 do   // http://192.168.13.12/control?cmd=event,HumidificationDemand=0
TaskValueSet,6,1,0
GPIO,15,0
//Mcpgpio,1,0
endon

on WaterLevelLowLow#State=1 do  // If water level rises and demand has been on.. turn back on 
if [HumidificationDemand#State]=1 and [HumidificationPumpStatus#State]=0 and [Pump_Temp#Temp]<110
GPIO,15,1
endif
endon

on Rules#Timer=1 do 
if [WaterLevelLowLow#State]=-1 or [WaterLevelLow#State]=-1 or [WaterLevelHigh#State]=-1
Reboot
timerSet,1,120
endif
endon

```

Humanoid-XXX, 2023-07-13:
I just realized my Rules were messed up.... MCPGPIO 1-3 should be input pullups and MCPGPIO 0 should be an output. 
Checking and will report back

tonhuisman, 2023-07-13:
> on WaterLevelLowLow#State!=1 do 
> if HumidificationPumpStatus#State=1 
> GPIO,15,0
> endif
> endon

This needs some square brackets too:
```
on WaterLevelLowLow#State!=1 do 
  if [HumidificationPumpStatus#State]=1 
    GPIO,15,0
  endif
endon
```

TD-er, 2023-07-13:
Running long wires for I2C will cause all kinds of issues.

These may pick up all kinds of noise, which may put the I2C device into some unknown state.

I2C can operate at really low frequencies, which is of course a great feature, but it may also cause issues like these.
For example if the clock line picks up some noise while it is addressing the MCP23017, the chip and the ESP may get out of sync regarding where they are in sending data.
For example the ESP may think it has sent 1 byte, but the MCP may think it has received 9 bits and thus is waiting for the next 7 bits.

Since the MCP is probably communicating a lot more compared to the temperature sensor, it is more likely any noise is affecting this chip.

A few things you could try:
- Used twisted wire, where you use a twisted pair for SDA + GND and one pair for SCL + GND
- Use an I2C multiplexer close to the ESP board. This will separate the I2C busses for both devices. (You need pull-up resistors on each I2C bus after the multiplexer and from the ESP to the multiplexer)
- Lower the pull-up resistance on the device furthest away from the ESP. This way the noise needs to have more energy to cause a logic level switch.

tonhuisman, 2023-08-05:
@Humanoid-XXX Did you make any progress in this issue, applying any of the tips provided?