letscontrolit/ESPEasy #4732: DCF77 Time (clock) signal transmitter

iz8mbw, 2023-07-13:
Hello.
I have Radio Controlled watches that uses the DCF77 signal.
Sometimes the DCF77 signal is weak so the watches cannot synchronize time.

There are some ESP32 and Raspberry Pi implementation to emulate the DCF77 signal using GPIO, so the DCF77 signal can be transmitted by the ESP32.

Here some links of implementation/codes:

https://github.com/SensorsIot/DCF77-Transmitter-for-ESP32
https://github.com/toto99303/Solo_DCF77_ESP32
https://blog.blinkenlight.net/experiments/dcf77/dcf77-generator/
https://wokwi.com/projects/347759046168674898
https://github.com/veebch/clock
https://github.com/hzeller/txtempus
https://github.com/elehobica/pico_dcf77_tx

Can it be implemented in ESPEasy?

Thanks!

TD-er, 2023-07-13:
I think it would not be too hard to implement, but the question is more like should we...
Or maybe even more relevant: "Are we allowed to?"

The DCF77 signal isn't that strong, even though it should have a range of 1500 km from Mainflingen.
So it isn't that hard to implement some simple transmitter, which will have quite a long range.
Just a rough estimate, I guess 20 mWatt (13 dBm) may already have a range of 1 km (assuming you have a somewhat properly tuned antenna)

Triggering a 77.5 kHz signal on a pin isn't that hard and you should then trigger short pulses of either 100 msec or 200 msec on another GPIO to change the amplitude of the 77.5 kHz signal.

However you also need to add some filtering or else you would cause all kinds of noise on higher frequencies as this will not generate a sine wave but a square wave.
Also the wavelength of 77.5 kHz is impractically long, so creating a properly tuned antenna may also be tricky.

In other words, I do see lots of opportunities to mess up the RF and start sending out a DCF signal which may be picked up by other DCF clocks in quite an unintended large area.

Dickesplumpel, 2025-11-10:
Hello, I think he meant something like this: "https://jurameer.de/index.php/elektronik/8-elektronik/48-dcf77-generator-2"
A dcf77-generator which gives dcf-code on a gpio-pin so you can replace a dcf77-receiver by esp with ntp, not sending over air. In our building-complex is a church with a motor-clock driven by a "müller dc 58.08 pro" which has an external dcf77 receiver but the time correction with dcf doesn't work. A ntp solution with esp would only be a nice to have, a toy.