letscontrolit/ESPEasy #4734: Ethernet not connecting in 20230623

V0JT48, 2023-07-15:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [x] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [x] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x] Tested previous/other build (20230508, 20230409, 20221224)
- [x] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

After updating to release 20230623 ethernet connection doesn't work using LAN8720. All previous versions tested work fine. It looks a bit  better in log after setting Power pin but connection still defaults to WiFi with Ethernet set as preferred.

264 : Info   : ETH PHY Type: LAN8710/LAN8720 PHY Addr: 1 Eth Clock mode: 50MHz APLL Inverted Output on GPIO17 MDC Pin: 23 MIO Pin: 18 Power Pin: 17
264 : Info   : ETH power ON
665 : Info   : ethResetGPIOpins()
773 : Info   : Set Network mode: WiFi

### Expected behavior
All works correctly in ESP_Easy_mega_20230508_normal_ESP32_4M316k_ETH and older versions:
267 : Info   : ETH PHY Type: LAN8710/LAN8720 PHY Addr: 1 Eth Clock mode: 50MHz APLL Inverted Output on GPIO17 MDC Pin: 23 MIO Pin: 18 Power Pin: -1
268 : Info   : ethResetGPIOpins()
1902 : Info   : ETH event: Started
1904 : Info   : ETH event: Connected
1909 : Info   : ETH event: Got IP
1952 : Info   : Webserver: start
1953 : Info   : processEthernetConnected()
1955 : Info   : ETH MAC: 8C:AA:B5:85:B0:C3 DHCP IP: 10.11.1.192 (ESP-Easy) GW: 10.11.1.1 SN: 255.255.255.0 DNS: 10.11.1.1/0.0.0.0 FULL_DUPLEX 100Mbps   duration: 4 ms
1957 : Info   : WIFI : Arduino wifi status: WL_NO_SHIELD 255 ESPeasy internal wifi status: DISCONNECTED
1958 : Info   : UDP : Start listening on port 8266
1958 : Info   : firstLoopConnectionsEstablished
3286 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 237328 EthSpeedState Link Up Full Duplex 100Mbps ETH status: Conn. IP Init

### Actual behavior

264 : Info   : ETH PHY Type: LAN8710/LAN8720 PHY Addr: 1 Eth Clock mode: 50MHz APLL Inverted Output on GPIO17 MDC Pin: 23 MIO Pin: 18 Power Pin: -1
264 : Error  : ETH: Settings not correct!!!
265 : Info   : Set Network mode: WiFi

### Steps to reproduce

1. Update/Install ESP32 node to release 20230623
2. Reboot
3. Obserrve logs and connect LAN cable

Tested on multiple units.

### System configuration

Hardware: ESP-WROOM-32, LAN8720

ESP Easy version: ESP_Easy_mega_20230623_normal_ESP32_4M316k_ETH

ESP Easy settings: Default, only Ethernet configured to
LAN8720 PHY Addr: 1 Eth Clock mode: 50MHz APLL Inverted Output on GPIO17 MDC Pin: 23 MIO Pin: 18 Power Pin: none

### Log data
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12588
ho 0 tail 12 room 4
load:0x40080400,len:2916
entry 0x400805c4
▒U187 : Info   :

INIT : Booting version: ESP_Easy_mega_20230623_normal_ESP32_4M316k_ETH, (GitHub Actions) mega-20230623 (ESP32 SDK 4.4.5.230614)
189 : Info   : INIT : Free RAM:247196
190 : Info   : INIT : Soft Reboot #4 Last Action before Reboot: Intended Reboot: CommandReboot Last systime: 168 - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
191 : Info   : FS   : Mounting...
217 : Info   : FS   : Mount successful, used 13303 bytes of 290156
228 : Info   : CRC  : Settings CRC           ...OK
232 : Info   : ESPEasy console using ESPEasySerial
248 : Info   : CRC  : SecuritySettings CRC   ...OK
254 : Info   : ESPEasy console using ESPEasySerial
255 : Info   : INIT : SPI not enabled
259 : Info   : ESPEasy console using ESPEasySerial
259 : Info   : INIT : Free RAM:241852
262 : Info   : ESPEasy console using ESPEasySerial
263 : Info   : INFO : Plugins: 48 [Normal] (ESP32 SDK 4.4.5.230614)
264 : Info   : ETH PHY Type: LAN8710/LAN8720 PHY Addr: 1 Eth Clock mode: 50MHz APLL Inverted Output on GPIO17 MDC Pin: 23 MIO Pin: 18 Power Pin: -1
264 : Error  : ETH: Settings not correct!!!
265 : Info   : Set Network mode: WiFi
366 : Info   : WIFI : Set WiFi to STA
471 : Info   : WiFi : Event STA Started
574 : Error  : WIFI : No valid wifi settings
576 : Info   : WiFi : Start network scan all channels
6790 : Info   : WiFi : Scan finished, found: 3
6793 : Info   : WiFi : WiFiConnected(), start AP
6895 : Info   : WIFI : Set WiFi to AP+STA
6899 : Info   : WiFi : Event AP Started
7461 : Info   : WIFI : AP Mode ssid will be ESP-Easy with address 192.168.4.1 ch: 1
7465 : Info   : WiFi : Event AP Stopped
7466 : Info   : WiFi : Event AP Started
7467 : Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
7472 : Info   : Webserver: start
8823 : Info   : WD   : Uptime 0 ConnectFailures 0 FreeMem 188628 WiFiStatus WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
13194 : Info   : AP Mode: Client connected: F8:16:54:CA:A4:40 Connected devices: 1
```

V0JT48, 2023-07-15:
It seems Power Pin has to be set to some GPIO, - None - setting doesn't work anymore.

TD-er, 2023-07-15:
This has been fixed in a recent commit...
The easiest work-around right now (till there is a new build) is to add some unused pin as power pin for Ethernet.

The settings check is too strict, so "not set" for that pin is considered to be invalid settings.

closed 2023-09-02 by TD-er