letscontrolit/ESPEasy #4740: Rules Editor - option for turn off the code completion

burdal1, 2023-07-19:
Hi, I would like to ask you to add the code completion on/off settings. The new editor makes me uncomfortable. The aggressive code completion makes it impossible for me to write the rules efficiently. I'm currently blocking espeasy.min.js, but that's not very comfortable.

I wonder if I'm alone with this experience.

Thank you 
Lukas

TD-er, 2023-07-20:
Sure, I thought it was already present as a checkbox to enable/disable the color-coding/code-completion script, but apparently not.

tonhuisman, 2023-07-20:
@burdal1 I've added a PR that introduces a setting for disabling Rules code-completion (and syntax highlighting). In a short while the build will be available from [this GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/5615576630), so if you can validate this works for you, that would be helpful. TIA.

burdal1, 2023-07-20:
Absolutely perfect! Tested on ESP_Easy_mega_20230720_normal_ESP8266_4M1M and it works.

Thanks You

PR #4742 (2023-07-20): [UI] Add option for disabling Rules auto-completion
Resolves #4740 

Features:
- Add Tools/Advanced setting to optionally disable the Rules auto-completion (when available). Will also disable Rules syntax highlighting!
Settings screenshot (in dark mode):
[image]

- Update documentation
closed 2023-07-20 by TD-er