letscontrolit/ESPEasy #4743: P035 IR TX

3dalex07, 2023-07-21:
Hi
It is posible to set pin output to  default High level  ?
in hardware page Output High not making any change

tonhuisman, 2023-07-21:
The IR library turns the Led OFF during initialization, and that seems to be the Low pinstate.
Why would you want to turn the Led ON?

TD-er, 2023-07-21:
There is an event when tasks are started, so you could act on it from the rules.

3dalex07, 2023-07-21:
i need invert logic

3dalex07, 2023-07-21:
i want to connect esp directly to other device ( I remove CHQ1838 from other device and connect esp)

TD-er, 2023-07-21:
I guess it will be better then to invert the signal using a transistor or optocoupler. (which will also provide isolation)

tonhuisman, 2023-07-21:
@3dalex07 I've added a PR to enable inverted output.
Can you test once [this GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/5625801351) is finished, and report your findings here?

3dalex07, 2023-07-22:
Thank. i will test it on monday.

tonhuisman, 2023-07-22:
@3dalex07 I've done some code improvements on the plugin, mostly to conserve some memory, and added documentation for the plugin. Can you test [this GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/5630651021) please?

3dalex07, 2023-07-25:
I tested output with oscilloscope. it work great! big thanks

tonhuisman, 2023-07-25:
Great, thanks for testing 👍

PR #4744 (2023-07-25): [P035] Add Inverted output option, as supported by the IRsend library
Resolves #4743 

Features:
- Add option to invert the output signal. This is a feature supported by the `IRsend` library.

TODO:
- [x] Testing by requester ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/4743#issuecomment-1650273090))
- [x] Add Plugin documentation
closed 2023-07-25 by TD-er