letscontrolit/ESPEasy #4749: More than one Pulse-counter allowed on ESPEasy?

HomAlix99, 2023-07-26:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [x] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x] Tested previous/other build (mention which one already tested)
- [x] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I have two pulse-counters engaged on ESP8266 (Wemos D1 mini) with ESPEasy (Built: mega-20220427).
ESP_Easy_mega_20220427_normal_ESP8266_4M1M.bin

When Task 3 is enabled, the counter works for a few second, than it doesn't count anymore (it seems that the pulse counter goes to sleep). Both inputs are connected to S0 outputs of energy-counters.
Configuration:
Task Device                            GPIO
3      Generic - Pulse counter GPIO-0 (D3)
4      Generic - Pulse counter GPIO-12 (D6)

Reboot or powercycle doesn't change the behavior.
The GPIO 0 (D3) is okay when used as a normal digital input, so it seems to be a software issue.
Both gpios are externally pulled-up.

I didn't find any hint to explain this behavior, but knowing that pulsecounters have to be defined between task 1 and 4.
Is only one allowed? 

br.

Alex

### Expected behavior

Both pulse-counters should work fine

### Actual behavior

Only the p.-c. on Task 4 works persistent

### Steps to reproduce

1. Define two pules-counters
2. 
3. 

Yes

### System configuration

Hardware:

ESP Easy version: 
ESP_Easy_mega_20220427_normal_ESP8266_4M1M.zip

ESP Easy settings/screenshots: 
[image]
[image]

### Rules or log data

```
On S0_Grid#Count_Grid Do      // Reagiert auf Event vom S0-Kontakt (Eingangsenergiezähler)
if [S0_Grid#Time_Grid] > 0       // Wenn Trigger-Ereignis vorhanden
   Let,2,[S0_Grid#Time_Grid]
   Let,3,1800000/[VAR#2]
   Publish,/%sysname%/Eingangsleistung/power_in,[VAR#3#D.0]
   Let,10,0
Else
   Let,10,[var#10]+1
   if [VAR#10] > 40
      Publish,/%sysname%/Eingangsleistung/power_in,0
   Endif
   // LogEntry,'Var 10: [var#10]'
Endif
Endon

On S0_Load#Count_Load Do      // Reagiert auf Event vom S0-Kontakt (Eingangsenergiezähler)
if [S0_Load#Time_Load] > 0       // Wenn Trigger-Ereignis vorhanden
   Let,4,[S0_Load#Time_Load]
   Let,5,1800000/[VAR#4]
   Publish,/%sysname%/Ausgangsleistung/power_out,[VAR#5#D.0]
   Let,11,0
Else
   Let,11,[var#11]+1
   if [VAR#11] > 20
      Publish,/%sysname%/Ausgangsleistung/power_out,0
   Endif
   // LogEntry,'Var 11: [var#11]'
Endif
Endon

on MQTT#Connected do
   Publish,/%sysname%/IP,%ip%
endon

```

tonhuisman, 2023-07-26:
> pulsecounters have to be defined between task 1 and 4.

That former limitation has been **fixed quite some time ago**, you probably have been reading the outdated Wiki documentation, current documentation for [ESPEasy in general](https://espeasy.readthedocs.io/) is on ReadTheDocs, and specifically the pulse counter: [Generic - Pulse counter](https://espeasy.readthedocs.io/en/latest/Plugin/P003.html)

TD-er, 2023-07-26:
A few things to keep in mind regarding pulse counter...
GPIO-0 MUST be high during boot of the ESP.
However when you connect some pulse generator circuit to it, you have no control over the state of that pin during boot and thus the ESP may no longer boot when it (re)boots.

Also, since this pin MUST be high at boot, it is being pulled up.
Other pins may not, so this pin will probably act different compared to other pins.
See: https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp8266

On ESP8266 pulse counting is completely being done in software.
This means it will process an interrupt on every pulse, which may affect performance of other code running on the ESP.
On the other hand, when the ESP is processing blocking code, it may take a while before it can collect the interrupt recorded data from a pulse.
I notice you do act on events from the pulse counter and trying to send data.
Processing this may take some time, which can cause some delay in processing new pulses and thus some pulses can be missed.

The pulse counter plugin is completely different on ESP32, where it can use a hardware counter and thus doesn't experience most of these issues you see on ESP8266.

So when possible, try using an ESP32 for pulse counters, especially when using >1.

N.B. also take into account the "best pins to use" page for ESP32 before wiring up, as the ESP32 does have some GPIO which require special attention. (e.g. do NOT use GPIO-12 on ESP32 as input (!!))

HomAlix99, 2023-07-27:
Hello everyone,

 

after firmware update to the newerst fw 20230623 both pulse-counters work fine now. 

So the issue can be closed.

Thank you very much for support.

I’m aware of the high-condition of D3/GPIO-0 while boot phase. In my case there should be no problem, bcs. the time between two pulses is at least 8 -10 seconds and any boot will be done on site.

 

Br.

 

Alex

tonhuisman, 2023-07-27:
> So the issue can be closed.

You can close the issue yourself, please. 👍