letscontrolit/ESPEasy #4752: [Domoticz] Add compatibility switch for Domoticz 2023.2

TD-er, 2023-07-31:
See list of deprecated command notation for Domoticz: https://www.domoticz.com/wiki/Domoticz_API/JSON_URL%27s#Warning_Stable_2023.2_.28and_newer.29

tonhuisman, 2023-08-05:
I don't think we need a compatibility switch to achieve this, ESPEasy already uses the required 'new' command format `/json.htm?type=command&param=**command**`, and the actually used commands are not in the transition list, so there shouldn't be issues there.

More important is that since the Domoticz 2023.1 release there is no longer support for including `username` and `password` as url parameters, but instead the common web-format `http://<username>:<password>@server:port/json.htm` should be used. As the commands are sent via the standard `SendToHTTP`-like code, that already uses the correct format.

So the only issue is with all the `SendToHTTP` commands from rules where the `username` and `password` parameters are used, these will have to be manually adjusted 😮 that'll involve some work...
NB: I don't think we should silently try to upgrade any SendToHTTP command using a username and password parameter in the url, that will not really help users to fix their code.

BTW: I don't particularly like it that Domoticz has made the choice of deprecating and even removing these parameters (I'm a Domoticz user myself) as it is an **interface**, and interfaces should _never_ loose any of their attributes.
But obviously, that's debatable 🤔

TD-er, 2023-08-05:
> NB: I don't think we should silently try to upgrade any SendToHTTP command using a username and password parameter in the url, that will not really help users to fix their code.

The SendToHttp command itself does support both.
So I guess we could make a selection in the Domoticz HTTP controller to switch this formatting.